motioneye-project/motioneyeos #1120: On second and following boots, RaspberryPi fails to get DHCP Address

gregewing, 2017-09-07:
Hi,  just found this project and really interested to see what it can do.

I've successfully installed ( a few times now) and reliably get a problem with the Raspberry Pi ( a Model B 1+ ).  It boots up fine on the first attempt, I can log in to the webUI and via SSH and on the local console.  but when I reboot it it fails to et a DHCP Address from the network.  I've re-installed the base Raspian, and a handful of other OSs and the problem has not reproduced itself so I'm confident its not hardware or the network environment.

I even tried your process for assigning a static IP Address using /etc/config/static_ip.conf ( if memory serves) and after a reboot I still get nothing.

Please advise.

ccrisan, 2017-09-07:
Please **attach** boot logs corresponding to the two scenarios: the not-working DHCP and the static IP configuration.

gregewing, 2017-09-08:
[logging.zip](https://github.com/ccrisan/motioneyeos/files/1287835/logging.zip)

Files attached.  this is the entire contents of /var/log under both scenarios.

I gathered these by reformatting the SDCard, writing the latest Dev image of motioneyeOS (motioneyeos-raspberrypi-20170827.img) to the card then booting it in the RPi.  I did this before each log file capture.  so the files should be similar up to a point, probably about when eudev does its thing.

I noticed when doing this that the problem seems to be caused by some form of conflict with the USB Camera I'm using being plugged into a local port on the RPi.  I tested with the camera plugged in on a USB Hub and it booted up fine, got an IP Address from DHCP and all the services started properly.  so the failure scenario is booting ( first and subsequent boots ) with the camera plugged in to a USB Port on the RPI board.   

As I mentioned previously I've tested the hardware configuration with other OSs, and with the same camera plugged in all the time to the RPi and not had this problem.

I've installed other OSs using NOOBS to check boot compatibility and they all let me boot and get a DHCP Address with the camera plugged in locally.
Successful tests with each of the following:
Raspbian
Raspbian Lite
openELEC
RISCos

ccrisan, 2017-09-08:
I think I found the problem. It seems that it's half motionEyeOS's fault and half that of your hardware. The following line indicates some USB issues, just as you suggested.

For some reason, one of the three [udev](https://github.com/ccrisan/motioneyeos/blob/master/board/common/overlay/etc/init.d/S12udev#L15) initialization commands fails and the init process stops.

I have changed the udev script so that it always exists with status code 0 when starting a4d9089c7cff83f6f947d656c706767aa0d6d9ea.

Now as a temporary workaround, you could manually edit your file and add a `return 0` at the end of the `start` function.

gregewing, 2017-09-09:
Col, thanks.  I'll give it a whirl and let you know if results are not as expected.  I take it this will make its way into the next release ?

ccrisan, 2017-09-09:
Yes, it'll be part of the next release, but I have yet to have your confirmation that this fix actually works for you.

gregewing, 2017-09-10:
Yup tested it today and it works like a charm. Thanks. 

I may be in a position to help with your roadmapnitems (3G/4g and VPN). I'll be in touch...

ccrisan, 2017-09-11:
The issue can be closed then.

closed 2017-09-11 by ccrisan

gregewing, 2017-09-11:
YES!  Thanks for your support.  ☺