motioneye-project/motioneyeos #1124: No UI from outside??? Is it normal?

canonmasta, 2017-09-09:
Morning all.
First of all thank you for this huge project! 

I have all configured. Inside my net i can see the User Interface and all the options but when i connect from outside i can only see the image of the cam i have. No options, no user interface. Is it a normal behavior? i connect with my public ip:8081 and i see this 

[image]

Maybe outside you cannot see more, don't know.
Thanks in advance for your help.

ccrisan, 2017-09-09:
There's nothing special motionEye(OS) does for when it's accessed from "outside" or from "inside". If you'd access port 8081 from "inside", you'll see the same stream with no UI.

To summarize: port 80 is for the UI, while port 8081 is for plain MJPEG streaming. If you want your UI from the "outside", forward **to** port 80.

closed 2017-09-09 by ccrisan

canonmasta, 2017-09-09:
Great! Great! Great! thank you!

canonmasta, 2017-09-09:
Now i have UI from the outside! this is incredible! Thank you! (and now with a couple of Playstation Eye added... even better)

ccrisan, 2017-09-11:
Make sure you don't actually use port 80 on the outside, but change it to something random (for the sake of not being scanned and possibly hacked).

canonmasta, 2017-09-11:
You are right.
Thank you.