motioneye-project/motioneyeos #1129: post api call on motion detection

joehas, 2017-09-13:
Hi, I am trying to make a post api call whenever motion is detected (to turn on my internet enabled outside lights). I've created the api call and tested that it works but I can't seem to trigger it when motion is detected (I could be doing something wrong). I've been trying to add it as a web hook but when click apply it crashes the OS and I have to reboot my raspberry pi.

Any advice or help would be appreciated..

details of the API call below with sensitive information removed.

curl --request POST "https://eu-wap.tplinkcloud.com/?token=YOUR_TOKEN_HERE HTTP/1.1" \
  --data '{"method":"passthrough", "params": {"deviceId": "YOUR_DEVICE_ID_HERE", "requestData": "{\"system\":{\"set_relay_state\":{\"state\":1}}}" }}' \
  --header "Content-Type: application/json"

Many thanks

Joe

joehas, 2017-09-17:
[motionlogs.zip](https://github.com/ccrisan/motioneyeos/files/1308955/motionlogs.zip)

Uploaded logs.

Updated issue details...

Recently I’ve been trying a new project, I have my outside lights controlled by a TP-Link internet enabled plug, I’ve managed to create the API call to turn the plug on and off (therefore turning the lights on and off) . I was hoping to use this whenever motion was detected by my camera. I first tried using my python script but then found out we can’t install anything on the OS as it’s not a normal Linux build. I then built an Ubuntu version to test it but had issues adding the camera and adding the run command. I get the “error occurred, refreshing is recommended” message.
 
I’ve now installed a fresh build of your latest raspberry pi 3 build - Motioneye version 0.37 Motion Version 4.0.1+git37b3595 OS Version motionEyeOS 20170827 and get the same error message when I try to do anything with motion notifications.
 
Is there anything you can suggest to get my idea working? I’m not really sure of the best way to call the API from the OS.
 
This is the CURL request (desensitised) 
 
curl --request POST "https://eu-wap.tplinkcloud.com/?token=YOUR_TOKEN_HERE HTTP/1.1" \
  --data '{"method":"passthrough", "params": {"deviceId": "YOUR_DEVICEID_HERE", "requestData": "{\"system\":{\"set_relay_state\":{\"state\":1}}}" }}' \
  --header "Content-Type: application/json"
 
This is the python code I created to do the API call
 
import requests
import json
 
url = 'https://eu-wap.tplinkcloud.com/?token=YOUR_TOKEN_HERE’
joe = r'{"method":"passthrough", "params": {"deviceId": "YOUR_DEVIDE_ID_HERE", "requestData": "{\"system\":{\"set_relay_state\":{\"state\":1}}}" }}'
obj = json.loads(joe)
print(obj)
r = requests.post(url, joe)
print(r.text)
 
 
I’ve also attached the logs from the new build.
 
Any help you could provide me would be greatly appreciated 😊
@ccrisan as discussed

ccrisan, 2017-09-17:
This description is way too long and too complicated. The things are actually simple:
1. You provide (write) a script that does your thing.
2. You put it into some folder (e.g. /data), you make sure it's executable and that it does its thing.
3. Under Motion Notifications you enable Run A Command and you enter the full path to your command.

Whatever code you're pasting here does not bring any additional details and won't help us solving the issue. So did you follow all the 3 steps above?

joehas, 2017-09-18:
Hi @ccrisan

I've done a bit more work on this today and got a bit further. I did try what you suggest before but I had built my python script using requests which I needed to install and can't using the motioneyeos.

I've rewrote my python code and confirmed it works when I SSH in to my Pi. (yay) it sits inside /data as suggested

I've made my file executable  "chmod +x tplink.py"

I've then selected "run command" on the interface and referenced /data/tplink.py 
When I click apply I don't get any errors this time (yay)

It's still not working though, when I make some movement in front of the camera the light doesn't turn on, which makes me think the script hasn't run.

Attached are 3 images, 1 to show you the settings under motion and notifications, 1 to show you the motion.log and one to show you the python script running on my putty session.

[image]
[image]
[image]

joehas, 2017-09-18:
@ccrisan 
*Update*
I've got it working. It seemed that I had to put #!/usr/bin/python at the top of my code. Run a command now turns the lights on and I have a run end command turning them off.

Thanks for your help, I think i'll write this one up so others can follow in the future.

closed 2017-09-18 by joehas