motioneye-project/motioneyeos #1130: Samba server authentication fails after 20170827 upgrade

dshults, 2017-09-13:
Running a Raspberry Pi3 with a portable 500GB WD USB HD, plus a DS3231 rtc. No changes took place other than the on-line upgrade. Mapped drives to motionEyeOS Samba shares using smb:// from either OpenSUSE or Ubuntu receive "Enter Password" or "Don't have permission..." error. Disabling the Samba server gives read-only access. All other UI, SSH and FTP access seem to work fine.

Here is a small snippet from log.smbd when failing:

[2017/09/12 17:56:16.725292, 0] ../lib/util/charset/codepoints.c:16630(get_conv_handle)
dos charset 'CP850' unavailable - using ASCII
[2017/09/12 18:05:50.611571, 0] ../lib/util/become_daemon.c:124(daemon_ready)
**STATUS=daemon 'smbd' finished starting up and ready to serve connections**
[2017/09/12 18:05:50.653328, 0] ../source3/lib/util_sock.c:334(open_socket_in)
**open_socket_in(): socket() call failed: Address family not supported by protocol**
[2017/09/12 18:05:50.653491, 0] ../source3/smbd/server.c:1080(smbd_open_one_socket)
smbd_open_one_socket: open_socket_in: Address family not supported by protocol
[2017/09/12 18:05:50.693447, 0] ../source3/lib/util_sock.c:334(open_socket_in)
open_socket_in(): socket() call failed: Address family not supported by protocol
[2017/09/12 18:05:50.693601, 0] ../source3/smbd/server.c:1080(smbd_open_one_socket)
smbd_open_one_socket: open_socket_in: Address family not supported by protocol
[2017/09/12 18:06:32.571558, 0] ../lib/util/charset/codepoints.c:16630(get_conv_handle)
dos charset 'CP850' unavailable - using ASCII

Also noticed this in the dmesg.log on every reboot, but it's been there for over a year:
   **IPv6: Loaded, but administratively disabled, reboot required to enable**

ccrisan, 2017-09-14:
See the release notes of [20170827](https://github.com/ccrisan/motioneyeos/releases/tag/20170827). There's a big warning there indicating what you have to do if you upgraded from previous versions.

closed 2017-09-14 by ccrisan

dshults, 2017-09-14:
Thanks for the speedy and simple solution Calin!
All good now...

ccrisan, 2017-09-14:
Things have become terribly complicated with recent authentication changes :(