motioneye-project/motioneyeos #1132: File storage to a network share doesn't work after updating

axispiercesirius, 2017-09-14:
Testing the connection says 
`Accessing network share failed: 'NoneType' object has no attribute 'lower'!`

I am using a Raspberry Pi 2B with the latest firmware

ccrisan, 2017-09-16:
Please clear you browser's cache, refresh and try again. If the problem persists, please attach your motioneye.log file here.

ShodaPublicity, 2017-11-29:
I am also receiving this error. Let me know if you need anything else.

RaspberryPi 3
motionEye Version: 0.37.1
Motion Version: 4.0.1+git2a140fa
motionEyeOS: 20171008

File Storage Settings:
Storage Device: Network Share
Network Server: nasew.local
Share Name: FreeNAS
Root Directory: /PiSecure

[motioneye.log](https://github.com/ccrisan/motioneyeos/files/1512648/motioneye.log)

Jasonb11, 2017-12-21:
I am also having this error: 
I just deleted all of my log files and have a fresh motion log. Motioneyeos keeps giving errors in GUI when a change is made. Example: adding a duplicate network camera.

Here is my motioneye.log
[motioneye (1).log](https://github.com/ccrisan/motioneyeos/files/1577638/motioneye.1.log)

Jasonb11, 2017-12-21:
@axispiercesirius @ShodaPublicity did you get anywhere with this message yet?

ccrisan, 2017-12-21:
Guys it appears to be a bug in motionEye when username is absent. I'll make sure to fix it for the next release. In the meantime, a workaround for now would be to supply a username.

ccrisan, 2017-12-21:
https://github.com/ccrisan/motioneye/commit/aef68011a811b681187d820dc2121807f7a2e5c3 should fix this issue. It'll be part of the next release.

closed 2017-12-21 by ccrisan

ShodaPublicity, 2017-12-21:
Hey @ccrisan, I use credentials to access my NAS so a username and password is supplied but I still have this issue. I thought that I included these fields so you would know that I have them filled but I just noticed that I did not, sorry.

Jasonb11, 2017-12-22:
Ill give it a try.