motioneye-project/motioneyeos #1134: Low resolution, zoom out. High resolution zoom in.

norey, 2017-09-17:
I'm using the old 5mp camera module and I noticed that when I lower the resolution, the preview and recorded video is zoomed out, while the higher the resolution, the preview and video looks zoomed in. What is causing that?

800x600 - http://i.imgur.com/aOJ73Tf.jpg
1920x1080 - http://i.imgur.com/u9cXC69.jpg

motionEye Version | 0.37
Motion Version | 4.0.1+git37b3595
OS Version | motionEyeOS 20170827

This is on a raspberry pi 2

Thanks

SuneBielefeldt, 2017-09-17:
I experience something that looks like:
When I choose a single camera as a full screen, I only see a smaller image, but as I finish, the picture is displayed for half a second.

motionEye Version	0.37
Motion Version  4.0.1+git37b3595
OS Version motionEyeOS 20170827

[image]

ccrisan, 2017-09-17:
@norey that behaviour is not about *zooming* but it's rather the way the PI camera manages its view port under the two different modes (video mode and stills mode, as far as I know). These modes are changed at some resolution threshold and that's what you're experiencing. This is not a motion(Eye(OS)) thing, but rather a Raspberry PI camera thing.

SuneBielefeldt, 2017-09-17:
I use 10 cameras + 6 link to a friend's address remote from me, in total 16 cameras. My Motion Eye hub (Raspberry Pi 3) with Motion Eye Os release 20170827. 

But you brought me on the track. I had configured 2 layout rows without understanding the meaning. 
Now I know it's important for display in single frame mode, full screen, but seen inappropriate, or what is the idea of ​​this option?. I first thought it was a consequence of the pre-release.

[image]

ccrisan, 2017-09-18:
@SuneBielefeldt you're right, those prefs shouldn't affect the full-screen mode. I'll look into it and try to find a fix.

@norey sorry for hijacking your issue :)

norey, 2017-09-18:
@ccrisan No worries man :)
That looks to me like a bug in the pi camera. I don't understand the reason behind this behavior, do you think there is a solution for it? Never heard about this issue before, is it documented somewhere?

ccrisan, 2017-09-18:
@norey http://picamera.readthedocs.io/en/release-1.10/fov.html

Look for FoV (field of view) and you might be able to find more details on this topic.

norey, 2017-09-18:
Another thing, is there a rotation feature for captured videos? (e.g. delete the oldest when storage is low) or should I write something up?

ccrisan, 2017-09-18:
@norey not yet.

norey, 2017-09-18:
Ok cool, then a simple cron job will do, thanks.

tokenwizard, 2018-03-02:
Any update on the Layout issue for Grid View vs. Full screen you mentioned on this thread as well as #679?

I'm having the same issue.