motioneye-project/motioneyeos #1136: Use official toolchain for all Raspberry Pi builds

jasaw, 2017-09-18:
Current Raspberry Pi builds are using buildroot's toolchain, which is causing Raspberry Pi `vcdbg` to fail, because `vcdbg` is closed-source and provided as a binary. Other precompiled /opt/vc binaries including libraries in [RPi firmware](https://github.com/raspberrypi/firmware/) have the same problem.

In order to get the precompiled binaries to work, we need to use the [official RPi toolchain](https://github.com/raspberrypi/tools).

It is probably worth providing all the /opt/vc binaries, similar to Raspbian.

ccrisan, 2017-09-18:
It's a thingOS issue for sure.

jasaw, 2017-09-19:
I've created this issue in thingOS as https://github.com/ccrisan/thingos/issues/5

closed 2017-10-12 by jasaw