motioneye-project/motioneyeos #1138: Add TLS/SSL to FTP

FlyingPersian, 2017-09-19:
Hi
Is there a way to add TLS/SSL to the FTP upload? This is the error I get when testing the connection:

Accessing the upload service failed: 550 SSL/TLS required on the control channel!

ccrisan, 2017-09-20:
I honestly don't know.

RamblingGeekUK, 2018-09-21:
I was looking at this, what is the web server you are using.  I have found conf files which state the port but when I change it to 443 or try to add it, things break.  Knowing the Webserver would make this easier it looked like it could be something based on python - or am I way off?

ijoseph, 2020-08-08:
`motion` uses [`Libmicrohttpd`](https://www.gnu.org/software/libmicrohttpd/). 
I see  
> `webu_mhd_features_tls: SSL/TLS: disabled`

in logs when trying to enable [`stream_tls`](https://motion-project.github.io/motion_config.html#stream_tls), which seems to come from 
[here](https://github.com/Motion-Project/motion/blob/bb0d931256a69321315020e1172893e01f8d0c1c/src/webu.c#L1565), which seems to indicate that `motion` was not compiled with TLS support. 

Maybe the FTP uses something different, though - not sure if what you're referring to is part of `motion`, `motionEyes`, or `motioneyesos`.

starbasessd, 2020-08-08:
You should be able to use scp (secure copy) that uses ssh protocol.
You'd have to use a TLS Certificate to use tls/https and that would
probably increase the complexity quite a bit, and also puts another load on
an already busy SBC...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?