motioneye-project/motioneyeos #1139: motioneye crashes

FlyingPersian, 2017-09-19:
Hi
I just installed motioneyeos on my RPI 3 Model B. MEO keeps crashing though. On the one hand, only motion crashes. This becomes visible in the GUI when the picture of the camera disappears. The log supports this, as at the same time as the picture comes back, motion is started.

https://pastebin.com/sGjD48YN <-- this log is only from booting it up at some point. 

On the other hand, the entire RPI reboots as well randomly or sometimes when I try to do something on it (e.g. save a setting, connect via SSH, trying to downlaod log files, etc.). I have debug logging enabled, but the log doesn't suggest anything.

https://pastebin.com/SqeZndPF <-- this is the entire log since I've installed motioneyeos. It's a 4h log with multiple crashes of the OS and motion itself. I'm on the latest install on the website.

jasaw, 2017-09-19:
@FlyingPersian Can you please attach the motion.log file too?

FlyingPersian, 2017-09-20:
@jasaw I installed MEO new, part the problems seem to persist. Here the log file of the new install, though I have the same settings as before:

https://pastebin.com/mDFsTvYv

The file is waay to big, it has 478044 lines haha. But this part keeps repeating:

[1:ml1] [DBG] [VID] v4l2_next: 1) vid_source->pframe 0
[1:ml1] [DBG] [VID] v4l2_next: 2) vid_source->pframe 0
[1:ml1] [DBG] [VID] v4l2_next: 3) vid_source->pframe 1 vid_source->buf.index 1
[1:ml1] [DBG] [VID] v4l2_next: vid_source->buf.bytesused 1382400
[1:ml1] [DBG] [VID] v4l2_next: the_buffer index 1 Address (73b5c000)

jasaw, 2017-09-20:
@FlyingPersian I don't see anything wrong with the motion.log file that you put on pastebin. The v4l2_next lines that keep repeating is normal when you crank up the log level.

FlyingPersian, 2017-09-20:
@jasaw hm well it still keeps crashing a lot, especially when I'm doing something on it (a.k.a. in the admin account). And capturing motion isn't working either. Taking Pictures manually works, but when I wave my hand in the picture for 3-4s it doesn't catch it. I set the minimum frames that need to have change to 20, and the fps of the videos is set to 30.