motioneye-project/motioneyeos #1140: Pi Zero - don't work if lost wifi

MatzeP, 2017-09-19:
Hi,
in the last week I must shutdown my wifi network and in these time my Pi Zero (with the newest version of motioneye) stop working. After restart the Pi Zero the storage folder has no data for these time.
Is that so correct?

jasaw, 2017-09-20:
Can you please be more specific about the actual problem you are having? When you say "stop working", do you mean it stops streaming or stops recording? What happened after you re-enable your wifi network? Did the Pi Zero recover itself and started streaming without user intervention?
What kind of set up do you have (fast network camera or not)?

MatzeP, 2017-09-20:
Hi,
I have configure that the camera should take every minute a picture. With existing network it was working fine, but without network no picture was make.

It is the standard raspberry pi camera.

After re-enable the Network he anwser and take pictures. Only the time beetween there are no pictures.

ccrisan, 2017-09-20:
@MatzeP motionEyeOS must have a network connection, unless you configure it so that [it runs in networkless mode](https://github.com/ccrisan/motioneyeos/wiki/FAQ#how-can-i-run-motioneyeos-without-a-network-connection).

If your network connection drops for some reason, motionEyeOS will reboot immediately in hope that the connection can be reestablished. See also https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting#system-continuously-rebooting.

So yes, it is expected that your system won't record anything if your network is down. Keep in mind that without a network connection, your Raspberry PI won't have a correct system date/time which is not really cool when it comes to video surveillance.

closed 2017-09-20 by ccrisan

juanjito88, 2019-12-13:
Is there any way to start continuous recording without relying on any network connection?

pdisner, 2019-12-13:
You mean as in a continuos loop, meaning your memory storage deletes the old to record the new? I’m sure there is. What are you using it for?

Thank You,
Paul Disner
(520) 307-6844
pdisner@me.com

juanjito88, 2019-12-13:
> You mean as in a continuos loop, meaning your memory storage deletes the old to record the new? I’m sure there is. What are you using it for? Thank You, Paul Disner (520) 307-6844 pdisner@me.com
> […](#)

Good morning, thank you very much and sorry for what I translate with Google from Spanish.
What I want is that when I start the raspberry, I start recording continuous video but without being connected to the network.
In the os.conf file, I set OS_NETWORKLESS to true, and in the advanced motioeye options I deactivate the LINK_WATCH option.
but when rebooting no longer makes any recording.
to see if someone can help me to configure that just turn on the raspberry start the video recording.
By the way, congratulations to the author of the OS, because it makes everything easier for those of us who don't have so much knowledge.
thank you very much