motioneye-project/motioneyeos #1141: Network share option not working - manual mount works ok

tommybuck, 2017-09-20:
Hi,

I'm trying to use the network share option for file storage. But it keeps failing to mount the share. I know the share settings work as I can mount it using the mount.cif command in the shell using the below command:

mount.cifs //192.168.1.105/studio1webcam /data/media/webcam/ -o "username=XXXXXX,password=XXXXXX,vers=3.0"

Can I edit the fstab file and use that way to mount it to a local folder then specify that in the file storage section?

ccrisan, 2017-09-20:
What's with the `vers=3.0`? Where did you get that from?

In theory you could use the [fstab.user](https://github.com/ccrisan/thingos/wiki/fstab.user) file to mount stuff at startup, but it probably won't work given that network won't be accessible at that time.

Another solution would be to run your `mount.cifs` command at [startup](https://github.com/ccrisan/motioneyeos/wiki/Tweaks#adding-custom-startup-commands), adding it to `userinit.sh`.

tommybuck, 2017-09-20:
the `vers=3.0 `tells cifs to use the smb version 3 protocol otherwise it uses smb 1 (very bad - wannacry uses it to spread!). See the mount.cifs man page.

I have put it in the userinit.sh script file and it works perfectly!

Shame it doesn't work in the web gui?

ccrisan, 2017-09-21:
@tommybuck does your command work without the `vers=3.0`?

tommybuck, 2017-09-21:
It will yeah but will use smb1 protocol by default. Not a good idea though as people are disabling smb1 on their windows machines. Also it will be disabled by default in the the next windows 10 feature update. Might be worth adding a drop down in the web gui that allows you to select the smb version???

ccrisan, 2017-09-21:
Leaving aside the `vers=3.0` part (which I will read about and see if can be used by default), the builtin code for mounting your share should also work. It uses pretty much the same syntax as yours.

Can you please try again mounting the share via motionEye (making sure the share isn't already mounted manually) and report back the motioneye.log file?