motioneye-project/motioneyeos #1142: No wifi motioneye

Winterhawk63, 2017-09-20:
Hello I have a Raspberry Pi2 B motioneye is installed on an sd card by itself. It works fine as long as I am using a network cable, but I can't connect by wireless. I am unable to see an ip address in my router when using the wifi dongle. Note when I swap sd cards to raspbian the wifi works fine. I see there is a preconfig topic in the motioneye help file, I have not done that at all. I watched many youtube videos and none showed having to reconfigure anything, they say plug in wifi and connect. So any help would be greatly appreciated. Can Motioneye and Raspbian be on the same sd card??
Thank you. John

ccrisan, 2017-09-20:
In addition to plugging in your WiFi adapter you must also configure it. motionEyeOS has no way of guessing your wifi credentials. If you have a working wired connection, just enable Network > Wireless Network and fill in the details.

If it doesn't work, see https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting#wireless-connection-problems.

Winterhawk63, 2017-09-20:
Thank you for the fast reply, also for all your work on the program. I have triple check my wifi credentials said and password, upper and lower case. I read through the help file but have checked out every thing that is there. No ip address shows up when I search it in my router.

ccrisan, 2017-09-20:
In that case, please attach your boot.log here.

Winterhawk63, 2017-09-21:
This boot log is when I was connected by network cable. I can’t access anything by wifi.
Thank you
John

Winterhawk63, 2017-09-21:
This is closed?? I sent my log file as you requested. I don't know if you received it or not? I attached the .txt file to my reply to you did I send it the wrong way?

ccrisan, 2017-09-21:
You probably attached it to the email, in which case it won't make it here. Please properly attach it to the github issue.

Winterhawk63, 2017-09-22:
At first I couldn't even get attaching this file to work, road blocks every step of the way lol. I hope this works and thank you for your help.

[boot.TXT](https://github.com/ccrisan/motioneyeos/files/1323091/boot.TXT)

ccrisan, 2017-09-22:
@Winterhawk63 your last boot log says:

     * Configuring wireless network: no device
     * Configuring wired network: dhcp

This indicates that there's no wifi adapter and that the wired connection is used.

Previous boots indicate that the wifi connection works properly:

     * Starting wpa_supplicant: done (rt2800usb)
     * Configuring wireless network: dhcp
    ...
     # Interface wlan0 has IP address 192.168.254.31/24

What seems to be the problem then?

Winterhawk63, 2017-09-22:
Yes the last boot log i was connected by a cable, that is thr only way i can connect to the Pi . My wifi adaptor works fine when using Raspbian.  Ok so when I'm trying to connect wirelessly in motioneye I can't see or find the ip address in my router. Without the ip address I can't connect to the Pi. Is it possible that my router isn't showing the ip address? It shows all the other ip addresses  on my network? Is there a better way to find the proper ip address? It sounds like everything is working I just need to access the Pi with the proper ip.

ccrisan, 2017-09-22:
Well the IP address is right there in your log file:

    Interface wlan0 has IP address 192.168.254.31/24

closed 2017-09-22 by ccrisan

Winterhawk63, 2017-09-22:
Ok one last thought, is it a browser issue. I know I had issues with Foscam loading in internet explorer and that is the browser I am using. But it works fine using the network cable.

Winterhawk63, 2017-09-22:
Thank you for all the help. I will do, that scan all ip addresses and try again. Funny that it doesn't show up in my router when it is working. Thank you again.

ccrisan, 2017-09-22:
@Winterhawk63 unfortunately routers aren't as wise as one would hope. Their UI often sucks and information is not always shown as expected. Take my Asus AC66U router for example, that simply won't list half of the clients connected and served via DHCP.

Winterhawk63, 2017-09-22:
Ok when I get home from work I'll have a chance to try a few things to get it working. I really appreciate the help, I know just enough about computers to be dangerous lol.

Winterhawk63, 2017-09-22:
Problem solved ... I put way to much trust in my router showing me the Pi ip address. Once I knew the proper ip address I connected right away. Thank you again for your patience and help.

jasaw, 2017-09-24:
I've updated wireless network troubleshooting guide in the wiki.

michaelquaye, 2018-05-03:
Can you assist me to capture motion and save on sd acrd when wifi is off with motioneye.