motioneye-project/motioneyeos #1143: Cannot add Raspberry Pi 3 remote camera

onlize, 2017-09-21:
Hello,

I configured camera module on RPi using the following instructions: http://elinux.org/RPi-Cam-Web-Interface

It works. I can go to http://RPiIPAddress/html and I can see everything through that camera. When I try to add it to my MotionEyeOs, I receive message "not a supported network camera". I tried to add this camera as Network Camera and as Simple MJPEG Camera, but have the same message.

What makes camera supported? I want to have simple camera added to MotionEyeOS software. How can I do it?

Thank you.

ccrisan, 2017-09-21:
What is the URL you're trying to add to motionEyeOS? If you try opening it in a browser, what does it look like? Screenshots are encouraged.

tommybuck, 2017-09-21:
@onlize try adding /cam_pic.php or /min.php on the end of your url. Does that work?

onlize, 2017-09-21:
The url is http://RPiIPAddress/html.

Here is what I see on the page (The camera is rotated):

[image]

onlize, 2017-09-21:
I do not care much about this software or any other. Is there any camera software that you would recommend? I want to have Raspbian with as little software installed as possible.

I have Raspberry Pi 3 with Camera Module attached.

Thank you.

onlize, 2017-09-21:
@tommybuck, I will try it later today when I come home. Thank you for suggestion.

ccrisan, 2017-09-21:
How about motionEye?

> I do not care much about this software or any other. Is there any camera software that you would recommend? I want to have Raspbian with as little software installed as possible.

onlize, 2017-09-21:
I will use MotionEye if I have to. I thought there was more basic solution.

ccrisan, 2017-09-21:
plain motion? [streamEye](https://github.com/ccrisan/streameye)?

closed 2017-09-21 by ccrisan