motioneye-project/motioneyeos #1144: Password for survellance issues

Siggkes, 2017-09-21:
Hi, love the software, thanks for your hard work. Using Raspberry Pi3.

When I put in the streaming link, with port forwarding etc, everything works fine. I can see my stream both on the local lan and on my PC and phone. Using the port : 8080 for example. When I turn on Authentication Mode and don't use a password, I get prompted and am still able to log on with just the username. If I then add a password to the surveillance account my browser rejects the combo. I have tried basic and Digest and reset the password many times. I am trying both on my pc on the local lan and from my phone so it isn't a cahce issue. What am I doing wrong? Have rebooted device etc. It seems to instantly accept my changes (ie, it will log on instantly if I remove password, or disable the need for credentials, but stops working as soon as I add password).

Siggkes, 2017-09-21:
Basically - If I remove the :8080 and just log in at the admin panel, the surveillance account credentials work. It's just if I use the streaming link. But the streaming link works fine with credentials removed or credentials on but the password removed.

ccrisan, 2017-09-21:
It's a known issue fixed in the latest version of motionEye. The fix will be part of the next stable motionEyeOS release.

closed 2017-09-21 by ccrisan

Siggkes, 2017-09-22:
Great, thanks. I downloaded the latest version yesterday, so does that mean it is not available yet?

ccrisan, 2017-09-22:
Yes, it means that it will be available soon.