motioneye-project/motioneyeos #1145: RPI Model B

danieljonasson123, 2017-09-22:
I installed motionEye yesterday with the new NOIR-camera, using the RPI model B, Is there any wiki on best settings for that model? Is it worth upgrading to a newer RPI lika RPI 2 or RPI 3?

ccrisan, 2017-09-22:
There's no such wiki and yes, it's worth upgrading to the RPi3.

closed 2017-09-22 by ccrisan

jasaw, 2017-09-22:
@danieljonasson123 Depends on what you're trying to do with your original RPi B. You can use it in fast network camera mode or motion detection and recording video.

1. If running fast network camera mode, the current version of motionEyeOS will work for you.
2. If doing motion detection and recording video on the pi itself, you will get low frame rate with current version of motionEyeOS. Improvements have been done but haven't made it into a release yet. In the meantime, if you want to try out the improved recording version, use my [forked version](https://github.com/jasaw/motioneyeos/releases/tag/20170827-drain-codec) to give you an idea of what the original Pi B is capable of, before you decide whether you need newer hardware.

As for the optimum configuration for the original pi B in option 2 (local recording mode), you can start with the below. Your main constraint will be the limited RAM.
- resolution: 800 x 600
- GPU memory: 66MB
- Pre-capture: 0 frame
- Frame rate: 20 fps
- Minimum motion frames: 15 frames or less
- Disable FTP server and/or Samba server (under web UI services section) if you don't need them.
- No live-stream and do not keep web browser connection open as it loads up the CPU.

danieljonasson123, 2017-09-25:
Many thanks for your inputs!