motioneye-project/motioneyeos #1146: HINT: Wifi issues with Raspi 3

nusspli, 2017-09-22:
Hi there,
its not an issue, its more a short hint.

I´ve had a lot of trouble with MotionEyeOS: low frame rate, disconnects, Raspi wasn´t found in network, motion detection was not working properly and so on.

It seems, that I found a solution: If you have an raspi 3, disable the onboard Wifi and Bluetooth and use a strong USB wifi dongle. I got a significant fps boost from 0,2 to 10-15 fps at 1024x786.

**How to setup MotionEyeOS?**
1. SSH into MotionEyeOS
2. enter:

> mount -o remount,rw /
> mount -o remount,rw /boot

3. nano /data/etc/modprobe.conf
4. insert the lines:

> #wifi
> blacklist brcmfmac
> blacklist brcmutil
> #bt
> blacklist btbcm
> blacklist hci_uart

5. reboot
6. Plugin your Dongle and wait a few seconds. Easy if you have had an Wifi configuration before as it automatically uses the same data.
7. done

What I don´t know right now ist, whether the energy saving mode (that also causes some onboard Wifi issues) is also deactivated. But I am pretty sure it is.

I hope you appreciate this hint and please don´t be angry  with me anymore as I´ve said some rude things because I was really upset about this unreliable behaviour (caused by the onboard Wifi)....

jasaw, 2017-09-25:
@nusspli Thank you for your suggestion. The Pi wifi signal performance is indeed weaker most dedicated wifi dongles. It would be very helpful to other users if you could also tell us the make and model of your wifi dongle.

Regarding the onboard wifi energy saving mode, can you please enter this command: `iwconfig` and checking for: “Power Management:off”.
If power management is off, then wifi energy saving is already disabled.

jasaw, 2017-11-14:
Closing issue due to lack of follow up.

closed 2017-11-14 by jasaw