motioneye-project/motioneyeos #1147: RPi2 poor performance with single rtsp stream

spxak1, 2017-09-23:
Hello,
I run an IP camera and feed that rtsp stream to MotioneyeOS on a RPi.

Running Motioneye on my Fedora Core 2 Quad server with the same feed works great, but I want the RPi to take over so that it runs 24/7/365.

Even with motion detection off, at 720p and 5fps, the feed on the web interface is very choppy. Sometimes the picture freezes for a good 20 seconds before it resumes. There is no way I can use this even as a simple monitor of motion.

Since so many people use MotioneyeOS, either I am doing something wrong, or for rtsp I need better hardware.

Would a RPi3 be significantly faster than the 2? I wouldn't think so given the similarities, but I am willing to buy one if I have to.

Any advice to get this working would be great.

Thanks in advance.

ccrisan, 2017-09-24:
The RPi3 is not significantly faster than a RPi2. I really doubt that the 20 seceonds freezes you experience on the RPi2 will go away when using a RPi3.

I recommend that you disable all functions (motion detection, still images, movies, webhooks, etc) and see if the problem persists.

Another idea is to make sure this is not a network issue by accessing motionEyeOS via a direct wired interface and having the RTSP stream on the wired network as well.

You could also replace your RTSP camera with a local USB camera at similar resolution & fps and see if the problem persists. If it does not, the RTSP stream poses problems and that's not a motionEyeOS issue.

spxak1, 2017-09-24:
Thanks.

I will try with a USB camera as the RTSP feed is wireless. 
At night, most pixels black, the feed is perfectly smooth. I think it might be a bandwidth issue. 

I will get back. Thanks again.

jasaw, 2017-11-14:
@spxak1 Any update on this issue?

spxak1, 2017-11-14:
@jasaw No, the RPi2 cannot do any better than that. With a local camera things are much better though, but for rtsp streams it just is not powerful enough to do the motion detection and save pictures.

I have tried a RPi3 and it seems to be coping much better, but nowhere near the smoothness of my PC. 

What does the job great, however, is the Odroid C2. That thing is a beast, but to me it is too precious for this task (I used it for home theatre running android).  

So I just use the PC (Fedora 26) to run motioneye and have feeds from the two rtsp streams + more when required. 

Thanks for asking.

jasaw, 2017-11-14:
@spxak1 Just out of curiosity, how many RTSP streams was your Pi handling? The RTSP streams are H264 encoded? Did you checked the CPU usage? Was it pegged at 100%? Which version of motionEyeOS was it on? Was the Pi configured to save movies into H264 format?

I'm just trying to understand where the performance issues are.

Chiny91, 2017-11-14:
I tried rtsp with a `Raspberry Pi 2 Model B Desktop (Quad Core CPU 900 MHz, 1 GB RAM, Linux)` says my Amazon order.   Looking at my notes of the time, my conclusion was `Experimented 28/12/15; not worth the cpu consumed for indiscernible pic betterment` but my recollection is that it was unusable as per @spxak1.   RTSP stream was H264.

grumpazoid, 2017-12-27:
I have  Pi3 and have spent a few days experimenting with a H264 stream from my DVR. 
The stream displays fine without on the web page any hangups. Pi and DVR are on a wired network.

One problem I have is that even if I set a working schedule or disable motion, video, stills etc. the pi still carries on asking for the rtsp stream and the motion daemon uses around 30% cpu. The ethernet led on the pi flashes like mad constantly as does the network led on my DVR. It would be nice if it only streamed when it actually needed it. Maybe this is so it is always available on the web interface?

I am going to try a USB camera to see if that makes any difference.

jasaw, 2018-01-14:
For anyone who is interested in hardware accelerated H264 decoding, please check out issue https://github.com/ccrisan/motioneyeos/issues/1228. No idea whether it's going to help with this issue.

jasaw, 2019-10-30:
Closing issue due to lack of follow up.

closed 2019-10-30 by jasaw