motioneye-project/motioneyeos #1148: Recorded video issues

norey, 2017-09-25:
I am having some issues with the recorded movies. They playback faster/slower than they are. Frames are either fast or slow in the same movie and not normal.

My video device is configured with 30 FPS and 800x600 resolution
Movie is H.264 (*.mp4) with 100% quality.

motionEye Version | 0.37
Motion Version | 4.0.1+git37b3595
OS Version | motionEyeOS 20170827

Any ideas?

jasaw, 2017-09-25:
Can you please try my forked version and report back your findings. https://github.com/jasaw/motioneyeos/releases/tag/20170827-drain-codec

norey, 2017-09-30:
I just had the time to install the forked system, and it does play recorded videos better. I assume those fixes will be available in the original system soon?

jasaw, 2017-10-01:
@norey Fantastic. Yes, all the patches are in motionEyeOS branch, and pending merge in upstream motion. I don't know which motionEyeOS release will incorporate these patches though.

jasaw, 2017-10-12:
All my patches are in pre-release 20171008.

closed 2017-10-12 by jasaw