motioneye-project/motioneyeos #1150: WebUI at HDMI

mecaz, 2017-09-25:
Hello, 

Im am using Raspberry PI for monitoring IP Cams on my network
And my RPI connected to a screen via HDMI. 
I can connect and monitor my cams connecting from my tablet or mobile phone
But i wonder is it possible to show my cams on HDMI like WebUI. 
It would be great if possible.
Regards

jasaw, 2017-09-26:
I think you're better off running Raspbian with PIXEL desktop environment. Follow the motionEye installation guide. As for showing the web UI on HDMI, auto launch chromium in kiosk mode, point it to localhost, and there you have it. You could also configure the mouse cursor to hide after some inactivity duration.

ccrisan, 2017-09-26:
This is not going to happen given that we do our best to conserve resources on the PI so that the actual video processing can be as good as possible.

closed 2017-09-26 by ccrisan