motioneye-project/motioneyeos #1151: Command triggered 'on_picture_save' doesn't work

superfino, 2017-09-25:
hi folks,
running motioneyeos 20170710 latest version but getting an error when i try to run a "Curl" 'on_picture_save'. This one runs well in a SSH terminal:
curl -k https://userID:password@IPaddress:port/xxxx?cmd=set%20RPIcam_Trigger%20on 
'on_picture_save' it caused the error 'curl: (6) Couldn't resolve host '2400n' in motion.log.
when i replace the %20 placeholder by spaces the error is curl: (6) Couldn't resolve host 'RPIcam_Trigger'
curl: (6) Couldn't resolve host 'on'.
possibly the %20 placeholders are interpreted as conversion secifiers but a space caracter also doesn't work. Anybody knows how to fix this problem?

ccrisan, 2017-09-26:
It may be that percents are not passed well from the UI to the config file and to the actual hook. I will take a look over this. Until then, i'd recommend:
 * trying to insert two percent signs %% where you'd like to have only one
 * enclosing the entire command into quotes, thus allowing the use of space character
 * building a simple bash script wrapper (e.g. `/data/myscript.sh`) that runs curl with required arguments and using it for on_picture_save.

ccrisan, 2017-09-26:
Yeap, your picture name pattern doesn't allow more than 1 picture per second to be saved. Add a %q somewhere.

ccrisan, 2017-09-27:
On a second thought, motionEye does what it can when it sees a percent - it simply places it in the motion config file. It's motion that interprets percents as format specifiers (or whatever they are called). So yes, the only correct solution here is to use two percents where you need one.

closed 2017-09-27 by ccrisan

superfino, 2017-09-28:
many thanks for your prompt and helpful response and for a brilliant surveillance tool.