motioneye-project/motioneyeos #1152: Support for EasyCap 4 Channel USB DVR

zzzbatmand, 2017-09-25:
I am aware that there is support for the "EasyCap USB Video Recorder" as shown in the issue #281, but is it possible to get support for the "EasyCap 4 Channel USB DVR" also?
I don't know if it is possible, but I would love it, if it is.

ccrisan, 2017-09-26:
@zzzbatmand what hardware are you on? Is this a Raspberry PI? If yes, have you tried Raspbian? Does it work?

zzzbatmand, 2017-09-30:
@ccrisan Yes, it is on a raspberry pi, sorry for not mentioning it.

I am not a professional at programming for raspberry and linux, but from what I have gathered from searching around, I have found that it is possible to use the device on raspbian, but it is problematic and not even certain that it will work.

From my perspective I think that it would be too hard to make it work, so this request can be put down if you don't want to work in it.

ccrisan, 2017-09-30:
If it doesn't work on Raspbian out-of-the-box then it won't work with motionEyeOS either, unless someone provides an easy way to make it work with the Raspbian kernel and no extra kernel modules. Till then I'm closing this issue.

closed 2017-09-30 by ccrisan