motioneye-project/motioneyeos #1153: Multiple cameras show single camera preview

sanchoooo, 2017-09-26:
I have a Pi 3b, 16Gb card. One camera is attached to the Pi 3 board, the other is a usb camera.  On the initial login screen, both camera previews show only 1 of the camera's images.
[image]

If I go into the setting of the camera not being shown, and change a setting there, both preview images then switch to that camera.  Disabling one camera shows the other, re-enabling the camera is again a crapshoot as to which one is displayed.

motionEye Version | 0.37.1
Motion Version | 4.0.1
OS Version | Linux 4.9.50-v7+

sanchoooo, 2017-09-26:
changed my os to this to see if I have the same issue.

motionEye Version | 0.36.1
Motion Version | 4.0.1+git37b3595
OS Version | motionEyeOS 20170710

on initial bringup and a quick test, both cameras were recognized and individual preview images were displayed.

benoit2600, 2021-07-05:
I know this issue is old, but I've got the same problem and found the solution : 

You need to set a different streaming port for each camera. 

Go to your 2nd camera, section Video Streaming, and set for exemple 8082, and that it.

closed 2021-07-05 by starbasessd