motioneye-project/motioneyeos #1155: Building image from source in Ubuntu 17.04 fails

paazmaya, 2017-09-28:
After building from sources successfully, the image creation fails, when using the latest  `master` at the time of writing this issue.

Used the command `./build.sh raspberrypi mkimage`

Full output:

```
:: creating boot loop device
20+0 records in
20+0 records out
20971520 bytes (21 MB, 20 MiB) copied, 0.0222222 s, 944 MB/s
:: creating boot filesystem
mkfs.fat 4.0 (2016-05-06)
:: mounting boot loop device
:: copying boot filesystem contents
:: unmounting boot filesystem
:: destroying boot loop device
:: creating root loop device
180+0 records in
180+0 records out
188743680 bytes (189 MB, 180 MiB) copied, 0.173248 s, 1.1 GB/s
:: creating root filesystem
mke2fs 1.43.4 (31-Jan-2017)
Discarding device blocks: done                            
Creating filesystem with 184320 1k blocks and 46184 inodes
Filesystem UUID: ab4ce28c-985f-490d-985e-a6c25232f114
Superblock backups stored on blocks: 
	8193, 24577, 40961, 57345, 73729

Allocating group tables: done                            
Writing inode tables: done                            
Creating journal (4096 blocks): done
Writing superblocks and filesystem accounting information: done 

tune2fs 1.43.4 (31-Jan-2017)
:: mounting root loop device
:: copying root filesystem contents
:: unmounting root filesystem
:: destroying root loop device
:: creating disk loop device
220+0 records in
220+0 records out
230686720 bytes (231 MB, 220 MiB) copied, 0.233066 s, 990 MB/s
:: partitioning disk

Welcome to fdisk (util-linux 2.29).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.

Device does not contain a recognised partition table.
Created a new DOS disklabel with disk identifier 0x51ace4c4.

Command (m for help): Created a new DOS disklabel with disk identifier 0x81c150d9.

Command (m for help): Partition type
   p   primary (0 primary, 0 extended, 4 free)
   e   extended (container for logical partitions)
Select (default p): Partition number (1-4, default 1): First sector (2048-450559, default 2048): Last sector, +sectors or +size{K,M,G,T,P} (2048-450559, default 450559): 
Created a new partition 1 of type 'Linux' and of size 20 MiB.

Command (m for help): Partition type
   p   primary (1 primary, 0 extended, 3 free)
   e   extended (container for logical partitions)
Select (default p): Partition number (2-4, default 2): First sector (43008-450559, default 43008): Last sector, +sectors or +size{K,M,G,T,P} (43008-450559, default 450559): 
Created a new partition 2 of type 'Linux' and of size 180 MiB.

Command (m for help): 

Command (m for help): Partition number (1,2, default 2): Partition type (type L to list all types): 
Changed type of partition 'Linux' to 'W95 FAT16 (LBA)'.

Command (m for help): Partition number (1,2, default 2): 
The bootable flag on partition 1 is enabled now.

Command (m for help): The partition table has been altered.
Calling ioctl() to re-read partition table.
Re-reading the partition table failed.: Invalid argument

The kernel still uses the old table. The new table will be used at the next reboot or after you run partprobe(8) or kpartx(8).

:: reading partition offsets
:: destroying disk loop device
:: creating boot loop device
/home/paazmaya/github/motioneyeos/board/raspberrypi/../common/mkimage.sh: line 148: * 512: syntax error: operand expected (error token is "* 512")
:: copying boot image
dd: writing to '/dev/loop10': No space left on device
1+0 records in
0+0 records out
0 bytes copied, 0.0200494 s, 0.0 kB/s
```

Anything additional I should check, such as versions?

ccrisan, 2017-09-28:
It may be that you have reached /dev/loop10 somehow. The `board/common/mkimage.sh` script apparently only works for /dev/loop0../dev/loop9. I recommend doing a reboot or manually getting rid of the first 10 loop devices.

jasaw, 2017-11-14:
@paazmaya Did the suggested method work for you?

paazmaya, 2017-11-28:
Hard to get recent Ubuntu to not have less than 10 loop devices, not sure where they pop up, but once I got the the loop devices used by the script to be with one digit, the image creation was successfull.

I tried pinpointing the place where this limitation comes from but ran out of time to do it further...

ccrisan, 2017-11-29:
I will make sure to fix the loop device limit.

ccrisan, 2018-04-23:
Fixed via https://github.com/ccrisan/motioneyeos/commit/2549027dce079e1794f57b8eb02fe17c659714d0.

closed 2018-04-23 by ccrisan