motioneye-project/motioneyeos #1157: Controlling Pan/Tilt servos

classofoutcasts, 2017-09-29:
Controlling my pan/tilt servos from python is very jittery, especially when my connected picam is set at higher resolutions. Suspect this is because the python GPIO library is using CPU cycles for timing.
There is the RPIO library using DMA instead of CPU for its timing which prevents this and provides very high timing precision (up to 1µs resolution).
I know it might be outside of the scope of this project to include all and every method for controlling servos, but any tips how to compile and add this library to Motioneyeos myself?
Or is it maybe worth for you considering to add this python lib to the OS as it is a very common purpose way of handling pan/tilt servos programmatically?

https://pythonhosted.org/RPIO/

ccrisan, 2017-09-30:
I think it actually makes sense to have this package integrated directly into [thingOS](https://github.com/ccrisan/thingos/) so other projects based on it can benefit of the package.

classofoutcasts, 2017-09-30:
I think it would benefit quite a few purposes, so thank you for considering it.
In the end I  tried to figure out other options as well and install pigpio in Rasbian.
This might be an even better library than RPIO as it it so more extensive and very userfriendly (not just python)

http://abyz.co.uk/rpi/pigpio/

timrowledge, 2017-12-02:
I've been trying to install pigpio on a motioneyeos system with not very much success.
Just for the record - and because I may be simply missing some stupidly obvious part - here is my attempt thus far.
a) download the pigpio package onto another Pi running full Raspbian.
b) make it, but don't run `make install`
c) use scp on the motioneyeos Pi to copy the compiled stuff 
'mkdir PIGPIO`
'cd PIGPIO`
`scp -r pi@192.168.1.X:/home/pi/pigpio/PIGPIO .'
d) since the install secrtion of the makefile wants to put files in /opt etc, 
`mount -o remount,rw /`
e) on the Raspbian Pi, 
`make -n install`
to list thre actual commands it wants to run
f) copy those lines and paste into the motioneyeos ssh window
g) curse the lack of `ldconfig`
h) use scp to copy /sbin/ldconfig* to /sbin
i) discover that running ldconfig complains about `/sbin/ldconfig.real: /lib/libstdc++.so.6.0.22-gdb.py is not an ELF file - it has the wrong magic bytes at the start`
j) give up because at this point I'm lost.

pigpio would be a very useful addition because it does a really good job of comtrolling the gpio, including servo control, sound generation, triggering from inputs and so on. It works well fro mC and Python and indeed Squeak.

ccrisan, 2017-12-03:
I will make sure to add `pigpio` to the next release.

classofoutcasts, 2017-12-03:
You really made my day! Can't use the great MotioneyeOS now as a I really depend on pigpio. When is the next release planned?

ccrisan, 2017-12-03:
I'll do my best to put together a stable release and a new testing release throughout this year :) That's the plan, but I'm not promising anything.

classofoutcasts, 2018-01-24:
No promises, I understand...
Do you reckon there's any chance pigpio will be available in MotionEyeOS soon?
Otherwise I need to think about other options, as my project's on halt momentarely waiting for this feature. No problem if that's the case, but I wouldn't want to put a lot of time in a custom solution when it's possible to use pigpio in MotionEyeOS shortly. Thanks for all the work you've already put in this project!

ccrisan, 2018-01-24:
Actually adding `pigpio` is not as simple as I thought it would be. It's not just a library written in C and/or Python - it contains a daemon that needs to be run at boot and continuously watches the GPIOs. I personally don't like this approach and I don't want polluting the OS with unnecessary processes.

I recommend using the two already available solutions for controlling the GPIO:
 
 * the RPi.GPIO python library
 * `/sys/class/gpio` Linux interface (also accessible via `gpio.sh` shell script, for convenience)

closed 2018-01-24 by ccrisan

classofoutcasts, 2018-01-24:
Sad to hear this. RPi.GPIO's output gets very unreliable and jittery when encoding or streaming streaming video simultaniously as it doesn't control the servo's realtime, but only when the cpu has clockcycles available for it. Pigpiod is a better implementation as it uses DMA for timing and therefore is completely unaffected by other (CPU consuming) processes on the system.

timrowledge, 2018-01-24:
No, actually it doesn't need to be run at boot. It can be started and stopped as needed with the usual `systemctl start pigpiod` type commands. Take a look at the /usr/bin/scratch script in plain Raspbian an see near the end where we check to see if pigpiod is already running and record that, start it for Scratch if not, run Scratch and then stop pigpiod after Scratch exits unless that ealrier check was positive.