motioneye-project/motioneyeos #1158: Raspberry PI Nano with ENC28J60 Modul Ethernet

jakespeed1311, 2017-09-30:
Hello,
i have downloaded  the image and on Pi version 3 ist runs wonderfull.
Now i have an Pi Nano with the SPI Card ENC28J60 and the system is in an reboot loop.
I have no change to login or have an cli to update the 
 /boot/config.txt with the driver file dtoverlay=enc28j60 or to enable spi if its not enabled.

What can i Do ? To geth this network card to work ?
tankx Jake

ccrisan, 2017-09-30:
motionEyeOS doesn't include any extra DTBs on the boot partition. You'll need to manually copy it from Raspbian, but keep in mind that the next upgrade will overwrite your boot partition, thus removing the file.

closed 2017-09-30 by ccrisan

jakespeed1311, 2017-10-01:
HI,
tank you for info. With Wlan i can help me now. This works great.
by
Jake