motioneye-project/motioneyeos #1160: Pi3/Pi2 + Camera v2 - Unable to open video device

montanoariel, 2017-10-01:
Hi

Idk if this is the right place to ask, but i was hoping someone will point me in the right direction. I've tried MotionEyeOS on a Pi2 and Pi3 with the same error

I've followed the guide posted in the wiki, however every time i complete the install and go check on the stream i m presented with the following error: "unable to open display"

[image]

I've tried debugging by running motion via an ssh session and I got the following output

`login as: admin
Welcome to meye-2bb0f761!
[root@meye-2bb0f761 ~]# motion
[0:motion] [ALR] [ALL] conf_load: could not open configfile /etc/motion/motion.conf: No such file or directory
[0:motion] [CRT] [ALL] conf_load: Not config file to process using default values
[0:motion] [NTC] [ALL] motion_startup: Using default log level (NTC) (6)
[0:motion] [NTC] [ALL] motion_startup: Motion 4.0.1+git37b3595 Started
[0:motion] [NTC] [ALL] motion_startup: Logging to syslog
[0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (NTC)
[0:motion] [NTC] [ENC] ffmpeg_init: ffmpeg libavcodec version 57.64.101 libavformat version 57.56.101
[0:motion] [NTC] [ALL] main: Camera 0 is device: /dev/video0 input -1
[0:motion] [NTC] [ALL] main: Stream port 0
[0:motion] [NTC] [ALL] main: Waiting for threads to finish, pid: 5572
[1:ml1] [NTC] [ALL] motion_init: Camera 0 started: motion detection Enabled
[1:ml1] [NTC] [VID] vid_v4lx_start: Using videodevice /dev/video0 and input -1
[1:ml1] [NTC] [VID] v4l2_get_capability:

cap.driver: "bm2835 mmal"
cap.card: "mmal service 16.1"
cap.bus_info: "platform:bcm2835-v4l2"
cap.capabilities=0x85200005

[1:ml1] [NTC] [VID] v4l2_get_capability: - VIDEO_CAPTURE
[1:ml1] [NTC] [VID] v4l2_get_capability: - VIDEO_OVERLAY
[1:ml1] [NTC] [VID] v4l2_get_capability: - READWRITE
[1:ml1] [NTC] [VID] v4l2_get_capability: - STREAMING
[1:ml1] [NTC] [VID] v4l2_select_input: name = "Camera 0", type 0x00000002, status 00000000
[1:ml1] [NTC] [VID] v4l2_select_input: - CAMERA
[1:ml1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
[1:ml1] [NTC] [VID] v4l2_do_set_pix_format: Testing palette YU12 (352x288)
[1:ml1] [ERR] [VID] v4l2_do_set_pix_format: Error setting pixel format.
VIDIOC_S_FMT: : Device or resource busy
[1:ml1] [NTC] [VID] v4l2_set_pix_format: Config palette index 17 (YU12) doesn't work.
[1:ml1] [NTC] [VID] v4l2_set_pix_format: Supported palettes:
[1:ml1] [NTC] [VID] v4l2_set_pix_format: (0) YU12 (Planar YUV 4:2:0)
[1:ml1] [NTC] [VID] v4l2_set_pix_format: 0 - Planar YUV 4:2:0 (compressed : 0) (0x32315559)
[1:ml1] [NTC] [VID] v4l2_set_pix_format: (1) YUYV (YUYV 4:2:2)
[1:ml1] [NTC] [VID] v4l2_set_pix_format: 1 - YUYV 4:2:2 (compressed : 0) (0x56595559)
[1:ml1] [NTC] [VID] v4l2_set_pix_format: (2) RGB3 (24-bit RGB 8-8-8)
[1:ml1] [NTC] [VID] v4l2_set_pix_format: 2 - 24-bit RGB 8-8-8 (compressed : 0) (0x33424752)
[1:ml1] [NTC] [VID] v4l2_set_pix_format: (3) JPEG (JFIF JPEG)
[1:ml1] [NTC] [VID] v4l2_set_pix_format: 3 - JFIF JPEG (compressed : 1) (0x4745504a)
[1:ml1] [NTC] [VID] v4l2_set_pix_format: (4) H264 (H.264)
[1:ml1] [NTC] [VID] v4l2_set_pix_format: 4 - H.264 (compressed : 1) (0x34363248)
[1:ml1] [NTC] [VID] v4l2_set_pix_format: (5) MJPG (Motion-JPEG)
[1:ml1] [NTC] [VID] v4l2_set_pix_format: 5 - Motion-JPEG (compressed : 1) (0x47504a4d)
[1:ml1] [NTC] [VID] v4l2_set_pix_format: (6) YVYU (YVYU 4:2:2)
[1:ml1] [NTC] [VID] v4l2_set_pix_format: 6 - YVYU 4:2:2 (compressed : 0) (0x55595659)
[1:ml1] [NTC] [VID] v4l2_set_pix_format: (7) VYUY (VYUY 4:2:2)
[1:ml1] [NTC] [VID] v4l2_set_pix_format: 7 - VYUY 4:2:2 (compressed : 0) (0x59555956)
[1:ml1] [NTC] [VID] v4l2_set_pix_format: (8) UYVY (UYVY 4:2:2)
[1:ml1] [NTC] [VID] v4l2_set_pix_format: 8 - UYVY 4:2:2 (compressed : 0) (0x59565955)
[1:ml1] [NTC] [VID] v4l2_set_pix_format: (9) NV12 (Y/CbCr 4:2:0)
[1:ml1] [NTC] [VID] v4l2_set_pix_format: 9 - Y/CbCr 4:2:0 (compressed : 0) (0x3231564e)
[1:ml1] [NTC] [VID] v4l2_set_pix_format: (10) BGR3 (24-bit BGR 8-8-8)
[1:ml1] [NTC] [VID] v4l2_set_pix_format: 10 - 24-bit BGR 8-8-8 (compressed : 0) (0x33524742)
[1:ml1] [NTC] [VID] v4l2_set_pix_format: (11) YV12 (Planar YVU 4:2:0)
[1:ml1] [NTC] [VID] v4l2_set_pix_format: 11 - Planar YVU 4:2:0 (compressed : 0) (0x32315659)
[1:ml1] [NTC] [VID] v4l2_set_pix_format: (12) NV21 (Y/CrCb 4:2:0)
[1:ml1] [NTC] [VID] v4l2_set_pix_format: 12 - Y/CrCb 4:2:0 (compressed : 0) (0x3132564e)
[1:ml1] [NTC] [VID] v4l2_set_pix_format: (13) BGR4 (32-bit BGRA/X 8-8-8-8)
[1:ml1] [NTC] [VID] v4l2_set_pix_format: 13 - 32-bit BGRA/X 8-8-8-8 (compressed : 0) (0x34524742)
[1:ml1] [NTC] [VID] v4l2_set_pix_format Selected palette YU12
[1:ml1] [NTC] [VID] v4l2_do_set_pix_format: Testing palette YU12 (352x288)
[1:ml1] [ERR] [VID] v4l2_do_set_pix_format: Error setting pixel format.
VIDIOC_S_FMT: : Device or resource busy
[1:ml1] [ERR] [VID] VIDIOC_TRY_FMT failed for format v4l2_set_pix_format: Device or resource busy
[1:ml1] [ERR] [VID] v4l2_set_pix_format: Unable to find a compatible palette format.
[1:ml1] [NTC] [VID] vid_v4lx_start: Using V4L1
[1:ml1] [NTC] [ALL] image_ring_resize: Resizing pre_capture buffer to 1 items`

motion appears to try and retry to set the palette , this then proceeds to stop and restart the process

any help would be appreciated

Thank you!

recursiveentropy, 2017-10-04:
This occurred for me as well, but only after having MotionEyeOS upgrade itself today (Oct 3, 2017). I was originally running v20170212, and, everything worked fine in v20170212, then after the upgrade today this camera error occurred ("Unable to open video device"). Note that Fast Network Camera works fine, but when I switch it off I get the same error.  

Further, if I delete the camera entirely from the MotionEyeOS config (via the gui) and try and re-add it, the "add" dialog shows that no camera exists to be added!

However, if I completely start from scratch by burning a new motioneyeos-raspberrypi2-20170710.img to my SD card, the camera is detected correctly and functions correctly.  It appears that something about the upgrade process is preventing the camera from being detected / functioning.  Perhaps the upgrade is corrupting the config somehow.

Power supply is sufficient, I'm using the same SD card for both scenarios above, and the same RPI/camera unit.

Raspberry PI2
Standard RPI camera (V4L2 camera)

TravelingTechGuy, 2017-10-09:
Similar issue:

I have a RPi Zero W with RPi Camera V2. I had it working well on 20170710. I recently updated (through the web app) to 20171003. I now have a blank (black) image. I removed the camera, and tried re-adding. The dialog shows "Local Camera"/"mmal service 16.1". The camera gets added, however the picture is still blank (the date and time on it are correct).

Is something in the latest  update not playing nice with the camera's driver? Should I go back to the previous version?

solarplex, 2017-10-22:
I have Camera V2 and Pi Zero and Pi Zero W. None of the local cameras show up on any of the newer versions.

jasaw, 2017-10-22:
@ccrisan Do you have a V2 Pi Camera to verify this problem? I haven't got one.

ccrisan, 2017-10-23:
@jasaw actually I only test the OS images with a V2 RPi Camera :) And so far, I haven't seen anything like OP's description.

solarplex, 2017-10-23:
My V2 Camera works fine with my Rpi3 but with my Pi Zero and Zero W, it doesn't recognize it. I've tested the last 3 builds and wondering if I need to build off of raspbian (which i didn't want to do).

vhaze, 2017-10-23:
here the same story, used the upgrade function on motioneyeos to upgrade to 20171003.  After a while, the upgrade process says succes, but the version number stays on 20170212, and all of a sudden, the local camera is not detected anymore.

`[root@v64a ~]# vcgencmd get_camera
supported=1 detected=1
[root@v64a ~]# modprobe bcm2835-v4l2
modprobe: can't change directory to '4.9.41-v7': No such file or directory
[root@v64a ~]# ls /dev/video*
ls: /dev/video*: No such file or directory`

is this solvable by remote ssh session?  the pi is in a very remote place, working ok and taking timelapse images for a few months now, but seems to be bricked now... 
changing the sd with a fresh install takes a lot of effort (high on a pole, while probably losing days of pics). i'd be able to give temp ssh access if desired.

ccrisan, 2017-10-23:
@vhaze in your case, you upgraded from a quite old version and the upgrade failed. You ended up with boot partition from one version and root partition from another. Your only chance is to rewrite the OS image from scratch.

solarplex, 2017-10-28:
Just for record, I have tested from the latest prerelease to 20170329. As well as Raspbian with MotionEye. As well as two different V2.1 Cameras with 2 different Pi Zero Camera Cables.

jolabpi, 2017-11-15:
Hi, got similar problems. Raspberry PI3 worked fine, PI camera and 2 Logitech c270 camera's then after a power failure and therefore a reboot  the Logitech camera's were not recognized anymore. Grey screen. I did everything. Reformatted SD card, reinstalled from scratch the latest image, used the old available original image, no success. Restored the old settings with the original image. No success. Deleting the cameras and trying to reinstall also does not work because the dialog shows "no loacal camera".
[messages.log](https://github.com/ccrisan/motioneyeos/files/1474845/messages.log)
[motion (1).log](https://github.com/ccrisan/motioneyeos/files/1474846/motion.1.log)
[motioneye (1).log](https://github.com/ccrisan/motioneyeos/files/1474847/motioneye.1.log)
Just now got an old Pi2 version and the light on the Logitech camera goes on. Unfortunately I need a Linux desktop to reset password because I have forgotten it.
Hope you can help because I like the program.
PS Pi camera works without problem.

jolabpi, 2017-11-15:
Update, everything works fine on old Pi2. Logitech camera is recognized and delivers image. 
On the Pi3 the MotionEyeOS image used to set up originally was 20160705. It works still fine with Pi camera, rtsp network stream camera but won't recognize the Logitech c270 webcams. I'm baffled.

jasaw, 2017-11-15:
Check whether the USB port on your pi3 is damaged.

jolabpi, 2017-11-16:
With a fresh head this morning and after reading your suggestion I removed everything from the usb ports ( had a keyboard and mouse connected ) plugged only the webcam in did a lsusb and dmesg |  tail command looked ok and everything works again. So my conclusion is, do not have anything else than the webcam(s) plugged into the usb ports when using MotionEyeOS. Thnx for pointing me in the right direction.

jolabpi, 2017-11-16:
Some further info. It turned out that one of my Pi3's was faulty. The current limiter (ap2553)  to the USB ports which supplies them with 5V was blocking. Strange enough after adding  "max_usb_current=1" to the config.txt file this de-blocked the current regulator. This command is used  by a  FET which changes the current limiting resistance by switching one resistor extra in parallel to the normal one.

ccrisan, 2017-12-16:
I think this issue can be closed and marked as invalid. Various different problems were reported in fact here but most of them have proven to be caused by some hardware fault.

closed 2017-12-16 by ccrisan

maltaresk, 2020-03-01:
i m getting error --  could not create component vc.ril.camera

starbasessd, 2020-03-01:
Need some more info:
Are you using a Pi3 or Pi2?
Which version (specifically) of motionEye & operating system or MotionEyeOS
are you using?
Can you use the camera in Raspbian?

https://projects.raspberrypi.org/en/projects/getting-started-with-picamera

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

maltaresk, 2020-03-01:
Thanks for responding. I am using RP 0W.. 
Itested on raspian, it's not detected. 
Concluding camera damaged, but why, I don't know. Even if new camera will have the similar fate. 
Can camera cable be a damaged link, I will try to get cable for pi zero, if available. 
Or something else?

starbasessd, 2020-03-01:
You turned the camera on in raspi-config? Make sure i2c is NOT on.
Make sure cable is oriented correctly in the PI and the camera.
There are 2 different cables that come with the camera, one is for the PiZ
and one is for all the other Pis.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

maltaresk, 2020-03-01:
No dear, camera not getting detected even in raspian.

starbasessd, 2020-03-01:
Make sure the blue tab faces away from the board on both the PI and the
camera.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?