motioneye-project/motioneyeos #1161: Building from source Documentation question

peterberbec, 2017-10-01:
First, thanks for the great setup here! It's a finely put together set of packages that are easy to install, customize and use. It's a fully finished product and you should be very proud! 

My question revolves around building from source. I successfully went through the build script and added nginx. I now have a reverse proxy setup and running, forcing HTTPS connections with a letsencrypt certificate. How would I "bake-in" configuration files? I know I can ssh in after the build, burn and boot-up to tweak the setup, but I was wondering if there was a method of saying "copy everything in /tmp/build/etc to /data/etc"?

Again, thanks for the great addition to the community.

peterberbec, 2017-10-01:
Second question. What happens to my custom build if I update through the UI? I assume it wipes my extra packages. Is there a method to install to the /data partition and symlinks them back? That's how I got htop,  bwm-ng and required libraries installed.

jasaw, 2017-10-02:
I don't know exactly how config files in /data/etc are managed, but I can answer your 2nd question.
When you check for firmware updates from the web UI, the default location is defined in /data/etc/os.conf file. The default values are:
```
os_firmware_method="github"
os_firmware_repo="ccrisan/motioneyeos"
```
You can change that to your own fork, or you can SSH in and upgrade by hand, which allows you to specify the firmware URL: `fwupdate upgrade <version|url>`

ccrisan, 2017-10-02:
@peterberbec your extra config files should probably go somewhere in `/usr/share/dataskel`. That directory is used to create the initial structure of the data partition right after it's formatted.

If you update through the UI you'll obviously lose your customized OS part, unless you do what @jasaw indicated - to provide builds for your own OS.

You may want to read more about how this OS works in [thingOS's wiki](https://github.com/ccrisan/thingos/wiki). It's far from complete, but you may get the idea of how things work in motionEyeOS.

closed 2017-10-02 by ccrisan