motioneye-project/motioneyeos #1164: Sys totally locks up if storage memory gets full Rpi3

Karlamees, 2017-10-03:
Has happened to me too many times already. Cant accsess from anywhere. Plugging Power wont help. Router wont show connection. Sys dead. Tired of reflashing OS all the time.

jasaw, 2017-10-04:
@ccrisan What do you think of separating the media storage to its own partition and end up with something like this:
- Boot
- Root filesystem
- Config partition (currently /data/etc)
- Media storage partition (currently /data/media)

This way, the device can still boot up as long as "Config" partition is not completely full. It doesn't matter if Media storage partition is full, just need to make sure motion software doesn't crash. There are things we can do to make sure we never fill up the "Config" partition.

ccrisan, 2017-10-04:
@jasaw having a fourth partition doesn't completely solve the issue and will only complicate the OS. There's nothing to prevent the user from changing the path to that partition, there are log files that could fill up the partition and there's no easy way to tell how to divide the remaining SD card space into the two partitions.

The correct way of solving the problem at hand is, in my opinion, to continuously monitor the available disk space and **automatically disable creating media files** or **remove older media files** before it's too late.

I know this is a highly requested feature but I haven't managed to find enough time to implement it yet. It has both a motionEye part as well as an OS-side part and has to be done carefully since we would be deleting user's files.

ccrisan, 2017-10-04:
By the way, this can be considered a duplicate of #785.

rahlzel, 2017-10-04:
...which is closed for some reason. 

I have also encountered this and been forced to reformat/reinstall MotioneyeOS as a result.

This is the correct fix:

> remove older media files

ccrisan, 2017-10-04:
The reason why it is currently closed is that I don't have time to deal with it. If anyone wants to take care of it and provide a PR, I'd be happy to reopen the issue and review the PR.

jasaw, 2017-10-04:
@ccrisan Regarding the separate media partition, I understand completely that it is difficult to repartition the disk now that you've got systems running in the wild. I feel that it would make it harder for an average user to lock up the system because they filled up the storage with their media files.

> There's nothing to prevent the user from changing the path to that partition

We could add a filter to motionEye to prevent users from using the "Config" partition for media storage. Users can still SSH in and tinker with the path, but they do so knowing very well that they're messing with the system, so all bets are off at this point.

> there are log files that could fill up the partition

We could use `logrotate` to manage the log files for us, making sure we never run out of space.

> there's no easy way to tell how to divide the remaining SD card space into the two partitions.

We could start with a 64MB partition. So we end up with:
```sh
BOOT_SIZE="20" # MB
ROOT_SIZE="180" # MB
CONFIG_SIZE="64" # MB
# The rest is media storage partition
```

For backwards compatibility, we mount the "Config" partition as `/data/etc`, and mount it as synchronous to reduce chance of filesystem corruption. We then mount the media storage partition as `/data/media`, and rbind it as `/home/ftp/sdcard` and `/home/ftp/storage`. Nothing else need to change.

PS: I have been using a more complex partition set up on all my projects, but the idea is the same: don't let media files fill up the root FS partition (and wherever the log files are) because nobody has time to monitor it. This has been working very well for me.

jasaw, 2017-10-11:
I've created PR #1172 to show what I'm talking about. Please comment in the PR thread.

jasaw, 2017-10-12:
Marking this as a bug as it is a serious issue that needs to be fixed.

ccrisan, 2017-10-13:
See https://github.com/ccrisan/motioneyeos/pull/1172#issuecomment-336370357 for a plan on how to prevent situations like this, in future motionEye versions.

DavHau, 2019-07-29:
I implemented a script that solves the problem by automatically deleting old files (Loop Recording). See my comment here: [#785(comment)](https://github.com/ccrisan/motioneyeos/issues/785#issuecomment-515941584)