motioneye-project/motioneyeos #1165: Passwords no longer being accepted after reboot. [20171003]

crw, 2017-10-03:
Steps to reproduce (describing actions taken on two RPi3s):

1. Installed 20171003 on two RPi3 devices.
2. Logged in as admin on web with previous password on both: success.
3. Logged in as admin on ssh with previous password on both: success.
4. RPi3 #1: rebooted to change a config setting (device name). After reboot, could no longer log in to web or ssh.
5. RPi #2: Changed password*. Rebooted. Could log in to web interface. Could no longer log in to SSH. 
6. RPi #2: Disabled SSH, rebooted. Could no longer log into web interface. 

* Changed to the same password. Since the password field only shows 5 dots, I thought this might be a problem. 

Is there any way to regain access to these machines without reinstalling from scratch? Do you know if this is an issue with the new release, and should I wait before re-installing?

Thanks!

ccrisan, 2017-10-04:
This looks like a duplicate of #1116. Although not clear yet, it might be caused by the use of an older (cached) version of the UI. If that is indeed the case, please try the following passwords:

 * `sha1("")`: `da39a3ee5e6b4b0d3255bfef95601890afd80709`
 * `sha1("null")`: `2be88ca4242c76e8253ac62474851065032d6833`
 * `sha1(<your admin password>)`

You can obtain the corresponding sha1 digest using the Linux command line (`echo -n "foo" | sha1sum`) or you could use http://www.sha1-online.com/.

Please let me know if one of the password variants above matches so we can be sure it's a cache problem.

ccrisan, 2017-10-06:
After further investigation, it looks like it's indeed a browser cache problem and the password is set to either `*****` or `133e3d6b775613651eaedbf7e609d244f2f852af`.

ccrisan, 2017-10-06:
And yes, this is indeed a duplicate of #1116.

closed 2017-10-06 by ccrisan

mrnoda, 2017-10-11:
Just experienced this issue, 133e3d6b775613651eaedbf7e609d244f2f852af worked fine for me.