motioneye-project/motioneyeos #1166: bug USB hard drive, sd card full

coME973, 2017-10-03:
Hello

I found a problem when the hard drive usb bug and that this is used to backup the videos.

When the usb disk is no longer recognized or mounted it is the sd card which is then used for recording.
the sd card is quickly full.
- 8 months ago: I see the SD card full, I reboot.
The system has never managed to restart, crash before the launch of services ssh, web .....
No linux machine available at this time, I was forced to reinstall motioneyeos.

- Today: sd card full, reboot without problem,
videos dump OK
but can not reassemble the usb drive.
totally abnormal file rights
I prefer to reinstall to have a clean system.

Here is a small problem, but motioneyeos is top :)

cordially

YG

rahlzel, 2017-10-04:
Looks like a possible duplicate of #1164

jasaw, 2017-10-11:
Yes, this is a duplicate.

closed 2017-10-11 by jasaw