motioneye-project/motioneyeos #1167: {"error": "internal server error"}

gryzli133, 2017-10-04:
Dear Ccrisan,

I have small Issue, since I'm on Holiday by my Parents, my MotionEyeOS on RPi3 is Hanging everyday after few Hour with following message from browser:
{"error": "internal server error"}
If I'm trying to reach RPi using Putty, then I get this message:
Network error: Software caused connection abort

Ping is possible anyway. A should say, that my parents have very bad Internet connection, using Internet Radio it is stopping few times every hour...

After reboot everything is working fine again.

Thanks in Advance for your Help!

ccrisan, 2017-10-04:
Please clear your browser's cache and refresh. If you suspect you have a problematic network, there's nothing we can do for you.

gryzli133, 2017-10-04:
The internet to outside is bad, but LAN inside is very stable, but even in the sam LAN there is this problem... Does Motion Eye OS need always an stable Internet to work stable? What can be the reason that even SSH is not possible?

ccrisan, 2017-10-04:
motionEyeOS requires a stable **local network connection**. Internet connection is used only for date/time, which is only used each 10-15 minutes.

TamirSlo, 2018-01-03:
I have got the same issue here.. bot due to network.. it happened to me yesterday when i left my RPi3 on for the night, when i waked up i had this error... I tried to reboot it and it failed as i checked on my laptop the sd card containing the OS was literally empty with no partitions... I reinstalled motioneyeos and it happened again... i havent tried to reboot yet nor checked the sd card but the issue is still active. If the pi will reboot this time i will attach logs here soon...

ccrisan, 2018-01-03:
@TamirSlo whenever you see such an error, there's a counterpart traceback in motioneye.log. We need to see that.

TamirSlo, 2018-01-03:
@ccrisan The thing is, I cant see the logs as somehow the sd card is formatted and I think the error relates to low power maybe...

ccrisan, 2018-01-03:
That may be a possibility. The new prerelease (20180101) has a new, improved way of reporting power issues. You may want to check it out.

TamirSlo, 2018-01-03:
I will try to update, but that is very weird.. Just today i checked for an update and it said i had the latest..? How come?

Thanks
Tamir

ccrisan, 2018-01-04:
@TamirSlo https://github.com/ccrisan/motioneyeos/wiki/FAQ#i-noticed-a-new-version-has-been-released-why-does-my-motioneyeos-say-its-up-to-date

TamirSlo, 2018-01-04:
Thanks, Anyway, i tested it tonight and there was no error but somehow the pi got turned off.. i had to go so i didn't have time to check it but I will see when i come back..

Thanks
Tamir