motioneye-project/motioneyeos #1168: Add WPA-2 key length as troubleshooting step to Wiki

recursiveentropy, 2017-10-04:
WiFi connections fail consistently with a WPA-2 hex key that is 64-digits in length, but 63-digits works fine. Recommend adding the below information to the troubleshooting list on this wiki page:
https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting#wireless-connection-problems

"**If your WiFi uses WPA-2 with a hexadecimal network key, make sure the key is no longer than 63 hex digits.  While hex WPA-2 keys can technically be up to 64-digits long, a hex key longer than 63 may cause a failure to connect.  This may show up during boot as a "wpa_supplicant: failed" error.  Try changing the key to a 63-digit hex value, or switch to an ascii passcode.** "

I've determined the above after a couple of hours of troubleshooting a failed connection with a RPI2 using the standard Edimax WiFi dongle.  After trying different various combinations of the factors below, the one clear root cause that emerged was the length of the WPA-2 key, if the key was in hex.  64-digits failed to connect, but 63 hex-digits succeeded. 

Test variables:
- Different WPA-2 configurations across various SSIDs;
- Swapping WiFi dongles;
- Using a powered hub; 
- Swapping power supplies;
- Swapping WiFi access points.

ccrisan, 2017-10-04:
I took your word for it and updated the troubleshooting wiki page as indicated. Thanks.

closed 2017-10-04 by ccrisan

recursiveentropy, 2017-10-04:
Cheers, thanks for the quick turnaround!