motioneye-project/motioneyeos #1174: Fast Network camera mode less efficient

Mikiya83, 2017-10-11:
Hi
I'm using MotioneyeOS on Rpi1 and Rpi3. Today i try to use "Fast Network Camera" on Rpi 1 and i'm suprised : in this mode, i get less FPS (with same resolution 1024x576) : ~2 fps instead of ~6fps in normal mode.
I can't understand why, CPU raise up to ~80% in normal and drop to ~30% in Fast Network Camera mode so i can't figure out where is the bottleneck :(
Do you have any ideas ?
Thanks

ccrisan, 2017-10-11:
How do you measure the FPS?

Mikiya83, 2017-10-11:
I use the metric displayed by motionEye on RPi3 (the RPi1 is added as Network cam).
And when i see the stream, it's clearly less smoth.
About the conf, i tested with more RAM for GPU (128 and 256), and i even tested in 640x480 and it's "stuck" around 2/3 fps :/

Mikiya83, 2017-10-16:
I retried with 20171008 but i still can get more than 2/3 fps in FNC, i do not understand why.

Mikiya83, 2017-10-17:
I'm sure i tested it a while ago and it worked fast, so i will check an older version to compare because it's a unsusable for now :(

Mikiya83, 2017-10-18:
Hum i can't understand why, but i reinstalled from scratch and it's ok now .... so you can close this issue, i can't figured out what happened sorry.

closed 2017-10-18 by ccrisan