motioneye-project/motioneyeos #1176: 20171003 drops WiFi after several minutes

glyndon, 2017-10-12:
After upgrading to 20171003, the Edimax USB WiFi stops working after a variable number of minutes (e.g. 10-60). If I reboot the Access Point it reconnects automatically, but again drops off-net after some time. Changing it to use a different (brand, model) Access Point made no difference.
Suspecting hardware, I traded it all out. RasPi 1B for a 2B (changing versions of MotionEyeOS to the one for the 2B) and a different Edimax adapter, even several different power supplies, including one that can source 5A. 
Same problem, no improvement.
Finally, I downgraded to MotionEyeOS 20170710, and all works fine again.
I'll be happy to supply whatever further info I can to help solve.

ccrisan, 2017-10-12:
Normally motionEyeOS will reboot itself if it senses a connection issue for one minute or so. What happens in your case? Can you connect a monitor and see what's going on? Or at least recover the log files?

glyndon, 2017-10-12:
I did check all the logs for anomalies, nothing seemed amiss. There was a recurring complaint about low voltage every 30 seconds, which is why I tried every 5V supply I had, but those never ceased. (I did see other threads about that issue, and it even happens on the older version which is operating stable, so presumed it's probably a red herring in this case)
Is there something specific I should look for, or a particular log to focus on?
I'll try it with a monitor on to see what the console reports, and post here.

glyndon, 2017-10-12:
I'm able to reproduce this:
Nothing unusual on the logs or console, except the repeating message about low supply voltage.
Everything works great as long as I keep the console 'awake' (by pressing keys, etc).
Once it blanks the console display (5 minutes idle), the WiFi goes away.

glyndon, 2017-10-12:
As best I can determine, it will work fine until I let the network idle (by not having a browser window open to it, and not pinging it) and then it shuts down after a few minutes.
Once it does, at the console I issue as root 
'ifconfig wlan0 down ; ifconfig wlan0 up' 
and it resumes normal function.
The command line tools indicate that power-saving is not active on the WLAN adapter.

glyndon, 2017-10-13:
Another observation:
When using 20170710, the indicator lamp on the WiFi adapter flashes with activity (normal for it).
When using 20171003, the indicator lamp is constantly on, never goes out, never flashes.

glyndon, 2017-10-13:
BTW, I think I found the reason for the continuous brownout warnings.
I (and presumably others) have used the /boot/config.txt settings to turn off the LED. The brownout init.d script spins on a loop looking at that LED's brightness, which is always 0 in such cases.

jasaw, 2017-10-13:
Going from version 20170710 to 20171003, I don't see changes that affect wifi, maybe except something in the kernel:

20170710: 
https://github.com/raspberrypi/linux/archive/04c8e47067d4873c584395e5cb260b4f170a99ea.tar.gz
20171003: 
https://github.com/raspberrypi/linux/archive/4153f509b449f1c1c816cf124c314975c3daa824.tar.gz

I know you said power saving is not active, but this might be worth a shot anyway.
I haven't tested this, just found it off the net.

> To turn off powersave, add the following to /etc/modules, or create a file (8192cu.conf) in /etc/modprobe.d/

```
# Disable power management
options 8192cu rtw_power_mgnt=0 rtw_enusbss=0
```

Verify with `cat /sys/module/8192cu/parameters/rtw_power_mgnt`

You'll need to remount root as read-write first.

ccrisan, 2017-10-13:
The module options are provided with the OS by default: [/etc/modprobe.d/modprobe.conf](https://github.com/ccrisan/motioneyeos/blob/dev/board/common/overlay/etc/modprobe.d/modprobe.conf). It might be worth however checking it with `cat /sys/module/8192cu/parameters/...` at runtime.

There's also a series of power mgmt disabling commands in the wifi startup script: [/etc/init.d/S35wifi](https://github.com/ccrisan/motioneyeos/blob/dev/board/common/overlay/etc/init.d/S35wifi#L86).

Either the kernel/rpi firmware/wifi firmware is recently suffering from a regression or the module or options have been renamed since kernel 4.4. @glyndon can you try running some tests with Raspbian? Does it exhibit the same behaviour?

glyndon, 2017-10-13:
I think you are pointing me in the right direction, and that this is all just a quirk in the new version's handling of that particular WiFi adapter. I appreciate your time and effort to help, especially since it's not your code that's at fault. (MotionEye is a fantastic bit of work, BTW) 
I'll keep pursuing it down the rtl8192c module path and see if I can find an alternative to it, and post any positive results here. (or I'll pull an ethernet wire to it and toss these adapters ;-)
For example: The module in this OS version is now named rtl8192cu, and seems to behave differently than its predecessor (the one without the rtl prefix). For example, the active /sys/....../parameters/ directory has no pseudo-file rtw_power_mgmt. 
I'm finding lots of threads where people are trying to solve this prob in Debian and related dist's.

ccrisan, 2017-10-13:
When they add the rtl prefix it usually means it has left the "staging" area and has been considered as a stable kernel driver. We should also update the modprobe.conf file accordingly. Can you please try and change the name of the module manually in that file?

glyndon, 2017-10-13:
if by that you mean changing
options 8192cu rtw_power_mgnt=0 rtw_enusbss=0
to
options rtl8192cu rtw_power_mgnt=0 rtw_enusbss=0

I had tried that, but during boot the 'rtl8192cu' module complains that it does not recognize those parameters.

glyndon, 2017-10-13:
Same hardware, same AccessPoint, no dropout problems under Raspbian 9.
And, FWIW, the indicator light on the WiFi adapter flashes with activity (as it usually does), rather than being solid-on as it's been when running the 10.03 vsn we're discussing here. 

...I wonder if the Raspbian rtl8192 drivers are portable to the OS underlying MotionEyeOS...

ccrisan, 2017-10-13:
@glyndon motionEyeOS uses exactly the same kernel (and drivers) that Raspbian uses. I'm downloading the latest Raspbian right now to run some tests myself.

ccrisan, 2017-10-13:
@glyndon my Raspbian uses the `8192cu` module and not the `rtl8192cu` one. I just downloaded the 2017-09-07 lite version. And yes, I'm using the same Edimax 7811 Un adapter.

How did you find out that you're using the `rtl8192cu` module?

ccrisan, 2017-10-13:
@glyndon I think I found and understood the problem here. The new Raspbian kernel enables both `8192cu` and `rtl8192cu` drivers, for some reason. One works and one doesn't. I never understood what's up with them. Anyways, `8192cu` works on the RPi while `rtl8192cu` works on x86/x64.

The latest Raspbian explicitly blacklists the `rtl8192cu` module, while this is not happening in motionEyeOS. I will make sure the next version blacklists it as well, but until then, you'll have to blacklist it yourself:

    echo "blacklist rtl8192cu" >> /data/etc/modprobe.conf

Then reboot and... fingers crossed.

glyndon, 2017-10-13:
So far, so good. 
To your previous question, I saw rtl8192cu in the report from lsmod.
It's now successfully blacklisted, not loading, and the LED on the WiFi adapter is blinking as it should be - a very good sign. 
I'll now leave it to idle for an hour or two without traffic and check it again.

glyndon, 2017-10-14:
Two hours of total [network] idle-ness later, still works.
I'd say your solution is a success!

Mikiya83, 2017-10-15:
Same problem here with 20171008, i think it's fixed in next release ? I just tried your "dirty" fix with echo until the "official" one, thanks !

ccrisan, 2017-10-16:
Fixed via https://github.com/ccrisan/thingos/commit/a9e266d6f5ba5638051d8187a01520031bb6986a. This will be part of the next release.

closed 2017-10-16 by ccrisan