motioneye-project/motioneyeos #1177: Endless reboot loop while "fetching interface information device not found"

chrisvel, 2017-10-13:
I restarted my RPI3 device and fell in a reboot loop. static_ip.conf has been setup and the RPI is connected to the network via ethernet cable. While it's loading the O/S after boot, I get a message "fetching interface information - device not found" and after 5 seconds it reboots. I don't know what happened, I haven't changed anything in the configuration - I just rebooted the device. Can anyone help ?

ccrisan, 2017-10-13:
I don't remember adding a message that says anything about "fetching interface information". Can you please provide a log file, a screen capture or even a photo of the monitor that displays that message?

chrisvel, 2017-10-13:
**boot.log**
```
---- booting motionEyeOS 20170710 ----
 * Detecting disk device: /dev/mmcblk0
 * Mounting filesystems: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Starting syslogd: done
 * Configuring CPU frequency: done
 * Starting eudev: done
 * Starting watchdog: done
 * Setting root password: done
 * Configuring wired network: no link
---- shutting down motionEyeOS 20170710 ----
 * Stopping motioneye: failed
 * Stopping smbd: failed
 * Stopping nmbd: failed
 * Stopping sshd: failed
 * Stopping crond: failed
 * Stopping date updater: done
 * Stopping network: done
 * Stopping eudev: done
 * Stopping syslogd: done
 * Detecting disk device: /dev/mmcblk0
---- booting motionEyeOS 20170710 ----
 * Detecting disk device: /dev/mmcblk0
 * Mounting filesystems: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Starting syslogd: done
 * Configuring CPU frequency: done
 * Starting eudev: done
 * Starting watchdog: done
 * Setting root password: done
 * Configuring wired network: no link
---- shutting down motionEyeOS 20170710 ----
 * Stopping motioneye: failed
 * Stopping smbd: failed
 * Stopping nmbd: failed
 * Stopping sshd: failed
 * Stopping crond: failed
 * Stopping date updater: done
 * Stopping network: done
 * Stopping eudev: done
 * Stopping syslogd: done
 * Detecting disk device: /dev/mmcblk0
```

**messages** log
```
Oct  6 13:28:45 motionEyeOS user.err motion: [3:nc3]Jan  1 02:00:14 motionEyeOS user.notice ethernet: no network connection available, rebooting
Jan  1 00:00:14 motionEyeOS daemon.info : starting pid 811, tty '': '/etc/init.d/rcK'
Jan  1 02:00:02 motionEyeOS auth.err passwd: no record of root in /etc/shadow, using /etc/passwd
Jan  1 02:00:02 motionEyeOS auth.err passwd: /etc/passwd: Read-only file system
Jan  1 02:00:02 motionEyeOS auth.err passwd: can't update password file /etc/passwd
Jan  1 02:00:02 motionEyeOS auth.err passwd: no record of admin in /etc/shadow, using /etc/passwd
Jan  1 02:00:02 motionEyeOS auth.err passwd: /etc/passwd: Read-only file system
Jan  1 02:00:02 motionEyeOS auth.err passwd: can't update password file /etc/passwd
Jan  1 02:00:12 motionEyeOS user.notice ethernet: no network connection available, rebooting
Jan  1 00:00:13 motionEyeOS daemon.info : starting pid 808, tty '': '/etc/init.d/rcK'
Jan  1 02:00:02 motionEyeOS auth.err passwd: no record of root in /etc/shadow, using /etc/passwd
Jan  1 02:00:02 motionEyeOS auth.err passwd: /etc/passwd: Read-only file system
Jan  1 02:00:02 motionEyeOS auth.err passwd: can't update password file /etc/passwd
Jan  1 02:00:02 motionEyeOS auth.err passwd: no record of admin in /etc/shadow, using /etc/passwd
Jan  1 02:00:02 motionEyeOS auth.err passwd: /etc/passwd: Read-only file system
Jan  1 02:00:02 motionEyeOS auth.err passwd: can't update password file /etc/passwd
Jan  1 02:00:12 motionEyeOS user.notice ethernet: no network connection available, rebooting
Jan  1 00:00:13 motionEyeOS daemon.info : starting pid 809, tty '': '/etc/init.d/rcK'
Jan  1 02:00:02 motionEyeOS auth.err passwd: no record of root in /etc/shadow, using /etc/passwd
Jan  1 02:00:02 motionEyeOS auth.err passwd: /etc/passwd: Read-only file system
Jan  1 02:00:02 motionEyeOS auth.err passwd: can't update password file /etc/passwd
Jan  1 02:00:02 motionEyeOS auth.err passwd: no record of admin in /etc/shadow, using /etc/passwd
Jan  1 02:00:02 motionEyeOS auth.err passwd: /etc/passwd: Read-only file system
Jan  1 02:00:02 motionEyeOS auth.err passwd: can't update password file /etc/passwd
Jan  1 02:00:13 motionEyeOS user.notice ethernet: no network connection available, rebooting
Jan  1 00:00:13 motionEyeOS daemon.info : starting pid 807, tty '': '/etc/init.d/rcK'
Jan  1 02:00:02 motionEyeOS auth.err passwd: /etc/passwd: Read-only file system
Jan  1 02:00:02 motionEyeOS auth.err passwd: can't update password file /etc/passwd
Jan  1 02:00:02 motionEyeOS auth.err passwd: /etc/passwd: Read-only file system
Jan  1 02:00:02 motionEyeOS auth.err passwd: can't update password file /etc/passwd
Jan  1 02:00:13 motionEyeOS user.notice ethernet: no network connection available, rebooting
Jan  1 00:00:13 motionEyeOS daemon.info : starting pid 811, tty '': '/etc/init.d/rcK'
Jan  1 02:00:02 motionEyeOS auth.err passwd: /etc/passwd: Read-only file system
Jan  1 02:00:02 motionEyeOS auth.err passwd: can't update password file /etc/passwd
Jan  1 02:00:02 motionEyeOS auth.err passwd: /etc/passwd: Read-only file system
Jan  1 02:00:02 motionEyeOS auth.err passwd: can't update password file /etc/passwd
Jan  1 02:00:12 motionEyeOS user.notice ethernet: no network connection available, rebooting
Jan  1 00:00:13 motionEyeOS daemon.info : starting pid 807, tty '': '/etc/init.d/rcK'
```
I will add a screenshot of the boot later. 

It seems strange that the date went from October to the beginning of epoch 1/1/1970.

ccrisan, 2017-10-13:
I see no "fetching interface information" message anywhere in your logs. I also don't see anything about a wireless connection, which leads me to believe that you haven't configured your wireless network. Have you? If yes, how?

ccrisan, 2017-10-13:
Scratch that, I had another wifi related issue in mind. You're not using WiFi connection at all in fact.

The `Configuring wired network: no link` message indicates that your wired connection doesn't actually work. Are you sure your cable is correctly, tightly connected at both ends? Have you tried Raspbian? Does it work?

chrisvel, 2017-10-16:
@ccrisan Sorry I couldn't get a better one: https://imgur.com/a/i0vLZ

Wireless is not used at all and wired connection has been setup and working for the last 3 months. 

I insist about the user's password that has been changed in the UI but wasn't changed in the o/s successfully.

ccrisan, 2017-10-16:
Hmm I believe this has to do with a problem with the USB controller (through which the Ethernet chip is connected to the PI). Does this happen always? If yes, it might be worth trying to replace the board, as it might be faulty. If it only happens randomly, it may also be a power supply problem.

chrisvel, 2017-10-16:
No, it was just happened after a reboot. I am going to re-image the SD card and try to see what happens. I'll update later.

petosak, 2018-06-08:
I have same problem on Pi 3B+
@chrisvel what helped you?

chrisvel, 2018-06-08:
@petosak I removed and pushed the same SD card in again, reimaged and restored the configuration. So far it has been working for months. I am not sure If it was the card (maybe some failure of the socket) or something else.

closed 2018-06-08 by chrisvel

petosak, 2018-06-08:
I thing, the trigger could be full SD card - maybe.
I cleaned some space offline, checked partitions and start again and same error.
OK. Let's start install again. Unfortunetely a have no backup.....

xtalker, 2018-06-20:
Same problem here, the big partition was 100% full.  I cleaned this up (using another Pi) but still won't boot, same error as mentioned above.

Boelen, 2018-08-06:
Can you guys look to this problem? Got the same problem now :).
I restarted my motion Eye OS & then it happened.
The connection is established after this error while its already rebooting...

BrutalCoding, 2018-08-08:
Same problem here as @Boelen

jschroer, 2018-08-11:
Same problem here, but on an odroid xu4

NathanKun, 2019-07-03:
For those who have the same issue, I found a solution.
I have a rpi 3b, wired to my router. One day the sdcard storage was full, after I did a reboot, it wouldn't boot anymore. It showed "Configurating wired network: up: error fetching interface information: Device not found" and kept rebooting.
I plugged the sdcard to a linux computer, by comparing the conf files with another functional rpi, I found the /etc/os.conf file is corrupted.

In my case the bad os.conf had only one line:
os_prereleases="false"

And the good os.conf has:
os_debug="false"
os_prereleases="false"
os_tty_login="tty1"
os_eth="eth0"
os_wlan="wlan0"
os_ppp="ppp0"
os_networkless="false"
os_country="GB"
os_firmware_method="github"
os_firmware_repo="ccrisan/motioneyeos"
os_firmware_username=""
os_firmware_password=""

By replacing the os.conf file, my rpi can boot correctly.
Hope this helps.