motioneye-project/motioneyeos #1178: starting wpa_supplicant: failed (brcmfmac)

plankton99, 2017-10-13:
getting this fail message on both Pi Zero W and Pi 3 B.
"starting wpa_supplicant: failed (brcmfmac)"  then it keeps rebooting.  

I have looked at all threads on this and there is no real answer, PLEASE HELP!

Checked all the boxes
1) diskpart clean and format
2) created wpa_supplicant.conf with unix EOL
3) doubled checked ssid and pwd spelling
4) ensured proper power delivered to Pi

I can't be the only one with this issue.

davimas, 2017-10-13:
Hey, i have this isue all half day long, nothing help.
Then i see in Notepad++ that the ending was .txt 😅

plankton99, 2017-10-14:
is there a way to load motioneye after the raspbian stretch desktop is installed with a s apt-get command?  Because the motioneye os is just not working

closed 2017-10-14 by plankton99