motioneye-project/motioneyeos #1179: Nextcloud as upload destination

escoand, 2017-10-13:
I've added a custom command to upload an image or movie to my nextcloud.

`curl -k -T %f -u nextcloud-upload-id: -H X-Requested-With:XMLHttpRequest https://nextcloud.example.com/public.php/webdav/%Y-%m-%d-%H-%M-%S.mp4`

Are you willing to add this as predefined Upload destination?

More info:
https://help.nextcloud.com/t/uploading-to-a-public-link-via-a-script/20294/5

ccrisan, 2017-10-14:
There aren't any plans to add support for NextCloud in the near future.

denics, 2022-12-22:
This feature has been merged now in https://github.com/motioneye-project/motioneye/issues/1999#issuecomment-1114854404 is there a remote possibility of getting it into motioneyeOS?

coder0107git, 2022-12-25:
@denics Until MotionEye version 0.43.0 is released I don't think any progress will be made on MotionEyeOS.