motioneye-project/motioneyeos #1180: Authentication Mode in streaming url don't work

bigpaolo, 2017-10-14:
Hy
i am now on motioneyeOs from motioneye and all works OK.
Only the Authentication Mode doesn't work for me.
If i choose basic or digest is the same.
i enter in http://myip:8081/ without Authentication Mode == WORK
i enter in http://myip:8081/ with Authentication Mode Basic, i insert my user and pasvord == DON'T WORK (show the promt for 2 times and then refuse te connection)
i enter in http://myip:8081/ with Authentication Mode Digest, i insert my user and pasvord == DON'T WORK (show the promt for 2 times and then refuse te connection)
I have try also https://... but is the same
I use user/password of the Surveillance and that one of the Admin but is the same
Where are stored user/password?

I use Raspberry pi3
I access from firefox, IE and edge but is the same

Can you help me please
Paolo

ccrisan, 2017-10-14:
What motionEye version do you run?

bigpaolo, 2017-10-14:
motionEye Version | 0.37.1
Motion Version | 4.0.1+git37b3595
OS Version | motionEyeOS 20171003

ccrisan, 2017-10-14:
Do you have any special characters in your Surveillance Password? Have you changed your Surveillance Username setting? If yes, have you used the changed username when trying to access the stream?

bigpaolo, 2017-10-15:
Hy
I resolve this night.
I have to change my username of surveilance mode to be able to connect with the basic autentication (with the default one i cannot be able to).

Thanks
Paolo

closed 2017-10-16 by ccrisan

toblery, 2018-11-23:
I just set my MotionEyeOS for Raspberry Pi Zero W and got it accessible from public internet. But unfortunately I didn't get Streaming authentication to work. 
So - I found reason for my problem with Chrome browser: https://www.chromestatus.com/feature/5669008342777856

So authentication via URL does not pass and now I don't get streaming visible. But I'm also protected for outsiders trying to find my camera stream.
Any idea how to make it work again? By using a cookie or something else? Try other browsers (on phone and computers)?

Thanks, Tobler

lbouriez, 2019-04-16:
Did you find any solution ?