motioneye-project/motioneyeos #1181: Upgraded to pre release bit does not reflect

b0rnwr0ng, 2017-10-14:
I upgrqded toghe latest prerelease but no longer see options and erion appears to be previous version.

jasaw, 2017-10-16:
Can you please attach your log files here?

b0rnwr0ng, 2017-10-18:
Sorry, I wound up reinstalling fresh.  I dont see options for h264 but
somethibg tells me Ibwould need to compile rather than use an image

jasaw, 2017-10-18:
@b0rnwr0ng You don't need to compile to get the H264 OMX encoder. It's already available in the [20171008](https://github.com/ccrisan/motioneyeos/releases/tag/20171008) pre-release.

rwijnhov, 2017-10-18:
Where do we enable the H264. Where do we find it in the ui?

jasaw, 2017-10-18:
Hardware accelerated video encoding (H264 OMX) is only available on Raspberry Pi hardware.

b0rnwr0ng, 2017-10-22:
I have a raspberry pi3 and do not see the option.

jasaw, 2017-10-23:
It's in the "Movie Format".

[image]

b0rnwr0ng, 2017-10-23:
Ahhh thank you.  I actually have been using fast vid camera and have that
disabled.  I will have to check it out though.  Thank you!

jasaw, 2017-11-14:
@b0rnwr0ng Issue resolved now?

b0rnwr0ng, 2017-11-18:
This can be closed.

closed 2017-11-18 by jasaw