motioneye-project/motioneyeos #1183: Please add support for HikVision cameras

Karlamees, 2017-10-16:
My friend just gave me an Hikvision Ip cam and i was really hoping that i could connect it to my existing MotionEye system.

ccrisan, 2017-10-16:
Find your HikVision camera streaming URL (RTSP or MJPEG) and it should work with motionEyeOS.

closed 2017-10-16 by ccrisan

Karlamees, 2017-10-16:
Im getting the error "Recv failure: Connection reset by peer"

ccrisan, 2017-10-16:
Again, make sure you're using the correct URL. Make sure the URL actually works well with Motion. If it does, it should work with motionEye(OS).

Karlamees, 2017-10-16:
i got the RTSP feed working on VLC using:
rtsp://<username>:<password>@<address>:<port>/Streaming/Channels/<id>/
But MotionEyeOS still says unsupported camera

ccrisan, 2017-10-17:
The following definitely is not a valid RTSP URL:
    rtsp://:@

    :/Streaming/Channels//`

Even if you remove the weird extra newlines, you get `rtsp://:@:/Streaming/Channels//` which is still not a valid RTSP URL, unless there's a syntax that I'm not aware of. Moreover, even if you replace `@` with the actual IP address, the two colons don't belong there.

Bottom line, make sure the URL works with Motion first. If it doesn't, there's nothing motionEye(OS) can do about it.