motioneye-project/motioneyeos #1184: KRACK update

Mikiya83, 2017-10-16:
Hi
Today KRACK attack against wifi wpa2 was publicly announced and some patches are already available. This is quite serious (more info here : https://www.krackattacks.com/) and debian is patched for some versions. So i just want to get some info, do you know when an update may be available for motioneyeos to include raspbian/debian's fix ?
Thanks !

ccrisan, 2017-10-17:
Judging from the [git log](https://git.busybox.net/buildroot/log/package/wpa_supplicant/wpa_supplicant.mk) of the wpa_supplicant in BR, this doesn't seem to have been addressed yet.

ccrisan, 2017-10-17:
Actually there are two pending patches that didn't make it to BR master yet (at the time of writing) but I'll make sure to apply them to thingOS/motionEyeOS asap:

 * http://patchwork.ozlabs.org/patch/826244/
 * http://patchwork.ozlabs.org/patch/826245/

Mikiya83, 2017-10-17:
Good to know thanks for the support 👍

Torxgewinde, 2017-10-19:
Hello!
It seems the patches are now available in Buildroot.

Thanks for your work!

Mikiya83, 2017-10-24:
It seems to be patched since this commit https://github.com/buildroot/buildroot/commit/57c0a485cc0a5681e772ddaf1c886e810d3d7ae4 indeed

ccrisan, 2017-10-29:
https://github.com/ccrisan/motioneyeos/commit/b1ec5ad89898fd16d9072c6698c1a51897be271e fixes this. It will be part of the next release.

closed 2017-10-29 by ccrisan