motioneye-project/motioneyeos #1189: Over the air updates not working.

AndyMAu, 2017-10-18:
I tried to update to latest version of OS by doing an over the air update.
I have 2 pi cameras (one rpi2 and one rpi3)
Both were unable to update from previous version ,so I had to download latest version and copy file to sd card.This was a bit tiresome,mainly due to having to access rpi unit and remove sd card .
The whole process of reloading settings is a bit painful.
Just so you know I even tried connecting rpi unit via small ethernet cable direct to router to perfom OTA update.Still no good.The device disappears from network and It's time to start from scratch.
Despite the time taken to reload ,there is one good thing I have noticed on this new update.I have a good solid network connection whereas before I had to keep rebooting raspi.
Thanks and keep up the good work.

jasaw, 2017-10-18:
Do you happen to have a log file saved somewhere that covers the remote firmware upgrade period?

AndyMAu, 2017-10-19:
Unfortunately,no.
As soon as I copied new version to sd card,any chance of saving any log files was gone.
New version very stable ,no network dropouts

jasaw, 2017-10-19:
@ccrisan I think one of the issues causing firmware upgrade failure is https://github.com/ccrisan/thingos/issues/8. If directories go missing, the device might not boot up anymore after that. We could add an rsync to make sure the directory structure is what motionEyeOS expects.
Thoughts?

ccrisan, 2017-10-23:
@jasaw you're right - that's likely the cause of this problem. It needs to be addressed as it may cause problems with future upgrades as well. Nevertheless it's not as urgent as others, since we're not planning on adding anything to the dataskel anytime soon.

I'll close this issue now as there's nothing we can do about it now.

closed 2017-10-23 by ccrisan