motioneye-project/motioneyeos #1190: Offline Mode

Freeze44, 2017-10-18:
Is there any solution to run motioneyeos without a network connection? The solution in the FAQ didn't work for me.

ccrisan, 2017-10-18:
The fact that "it didn't work" for you won't allow us to help you. You need to come with more information (log files, errors, what exactly did not work).

Freeze44, 2017-10-18:
When i setted the os_networkless variable to true i got the following error "can't stat /data/etc/os.conf: structure needs cleaning"
/etc/init.d/conf:line 15: /data/etc/os.conf: structure needs cleaning
I`m sorry but i`m a beginner and it`s my first raspberry-based project.

ccrisan, 2017-10-18:
@Freeze44 that sounds like a filesystem corruption problem.

jasaw, 2017-11-14:
@Freeze44 Have you tried reflashing your SD card?

jasaw, 2019-10-30:
Closing issue due to lack of follow up.

closed 2019-10-30 by jasaw