motioneye-project/motioneyeos #1192: Missing menu items in configuration menu

barjac, 2017-10-18:
Hi,
I just created two SD cards using the writeimg script and one has all options but the other has not.
sudo bin/writeimg -i /home/baz/arm/MotionPie/motioneyeos-raspberrypi-20171008.img -d /dev/sdg -s 192.168.1.99/24:192.168.1.253:8.8.8.8
[image]

The one without is on an 8GB Sandisk card, the one with is on a 4GB Transcend card - both on identical RPi 1 Bs with identical Rpi cameras.
I saw an earlier similar report where either the user had missed the Expert Settings or maybe it appeared on re-boot, so I tried re-booting, but no change.
I have both systems side-by-side in a browser and no way can I get more options in one of them.
Both are logged in as admin with passwords
I am tempted to just try again, but thought you may wish to debug this - I can let you have ssh access to the device as you did a couple of years ago for another bug. Let me know and I will set it up and email details.
Is there something I can check over ssh?
Barry

ccrisan, 2017-10-18:
Can i see a screenshot of the General Settings opened in both windows?

ccrisan, 2017-10-18:
:)

closed 2017-10-18 by ccrisan