motioneye-project/motioneyeos #1193: Periodically clear log files

mehtaman, 2017-10-19:
Hi,

Is there a way to clear the log files?
Or set up a task to clear them after 7 days for example.

Some of my log files are over 200mb and take a long time to load up in a text view.

Thank you
:-)

jasaw, 2017-10-19:
I thought we have logrotate already, and max log size is set to 10MB.

ccrisan, 2017-10-19:
We do have it and it should work. @mehtaman if that is not the case for you, please tell us the version of the OS you're using and a `ls -l /var/log`.

mehtaman, 2017-10-20:
Just as an example, my current motion.log is 252MB:

[image]

messages.log is around 69MB, motioneye.log 7MB and the others are below 1MB

I am using version: 
motionEye Version | 0.37.1
Motion Version | 4.0.1+git2a140fa
OS Version | motionEyeOS 20171008

[image]

jasaw, 2017-10-20:
I think I know what's going on.
Logrotate only runs once a day, at 3:14am everyday. If there are lots of messages in the day, the log file will go over 10MB and not rotate 3:14am. We need to run logrotate more often.

mehtaman, 2017-10-20:
Thank you, as a work around have you got the best commands I can run to:
1) Stop Logging service
2) Flush log folder of current log files
3) Start Logging service

My Linux is weak and I don't want to break anything :+1:

jasaw, 2017-10-20:
I am more concern about the really large log files you are getting. Did you enable debug from the web interface?

You don't need to worry about deleting the log files. Logrotate will delete the old files for you.

As a workaround for now, you could run logrotate every hour by adding this to your crontab:
`0 * * * * /usr/sbin/logrotate /etc/logrotate.conf`
More cron info on [wiki](https://github.com/ccrisan/motioneyeos/wiki/Tweaks#running-commands-at-regular-intervals-cron-support).

mehtaman, 2017-10-20:
@jasaw Debug was on until now... I'll see how the logs are with debug mode disabled.

Then give the CRON job a go if the files are still over 10MB tomorrow.

jasaw, 2018-02-28:
@ccrisan What do you think of running logrotate more than once a day?

ccrisan, 2018-02-28:
@jasaw I don't see any harm in that.

Chiny91, 2018-02-28:
Conventionally, logrotate is set to run once per day, although for logs that build slowly then the logrotate.conf directive `weekly` or `monthly` is used (presumably there is a default but I don't know what that is for motionEyeOS).   For normally fast building logs, which for motionEyeOS is more than 10 MB per day, then running logrotate more than once per day is the answer, although I'd want to know why the log is normally building so fast.

It is usually possible to force (use `-f`) logrotate; (not checked that exists for motionEyeOS) which is useful for testing.   Does logrotate actually work, or has some gremlin crept in ?

jasaw, 2018-03-01:
@Chiny91 The logs grow quickly when debug mode is enabled.
This is a 2nd line of defense against unexpected large log files, like users forgetting to turn off debug mode, or something in the system going wrong and spewing excessive stuff into log files.