motioneye-project/motioneyeos #1194: Sub-optimal Pi Zero W overclock settings

jasaw, 2017-10-20:
On a Pi Zero W, the overclock settings is slower than the factory default frequencies.
This is the current motionEye(OS) Pi Zero overclock settings:
```
arm_freq=1000
gpu_freq=300
sdram_freq=450
over_voltage=6
```

Everything is good except the `gpu_freq` underclocks the `core_freq` on a Pi Zero W.

### gpu_freq
Sets `core_freq`, `h264_freq`, `isp_freq`, and `v3d_freq` together. On Pi 1/Pi 2 the default value is 250 for all items, on Pi 3/Pi Zero /Pi Zero W `core_freq` defaults to 400 and `h264_freq`, `isp_freq` and `v3d_freq` default to 300.

### core_freq
Frequency of the GPU processor core in MHz. It has an impact on CPU performance because it drives the L2 cache and memory bus. The default value is 250 for the Pi 1/Pi 2 and 400 for the Pi 3 and Pi Zero and Pi Zero W. Note that the L2 cache benefits only the Pi Zero/Pi Zero W and Pi 1, but there is a small benefit for SDRAM on the Pi 2/Pi 3.

Full documentation [here](https://www.raspberrypi.org/documentation/configuration/config-txt/overclocking.md).

What would be nice is to have the "None" overclock option remove all the `arm_freq`, `gpu_freq`, `sdram_freq`, and `over_voltage` lines from /boot/config.txt, rather than setting them to some predefined values.

ccrisan, 2017-10-23:
Yeap, the best thing to do with "None" is to leave the fields unset. I'll take care of this.

ccrisan, 2017-10-31:
Fixed via https://github.com/ccrisan/motioneyeos/commit/c06442ccee2f8f5f8f3fe1f36b941be9ca8445fa.

closed 2017-10-31 by ccrisan

timdonovanuk, 2017-12-28:
Surely this is still an issue? This issue was not just about resetting "none" back to optimal settings for the PiZero, it was the fact that the overclocking option "PiZero" is incorrect. It sets the PiZero to 1000/300/450/6 but the "300" setting is lower than the default of "400" that the PiZero ships with.