motioneye-project/motioneyeos #1195: Having trouble accessing the webpage from WiFi?

Isaac96, 2017-10-21:
I have motioneyeOS 20171003 on a Pi 3. I have the Pi connected directly to my router, which is a small one I am using for this project only. I can access the stream from a PC connected to the router over LAN, but another PC connected using WiFi cannot access the stream. I can SSH to the Pi from the not-working PC over WiFi. The not-working PC cannot connect to Ethernet and the one which works has no Wifi card.
I have restarted the Pi, fully reset the router, and rebooted both PCs. Any ideas of what the problem could be?
Thanks!

jasaw, 2017-10-22:
> I can SSH to the Pi from the not-working PC over WiFi but cannot access the stream from WiFi.

What resolution and frame rate are you running? It is possible that your wifi can't handle the amount of data you're trying to push across it.

> The not-working PC cannot connect to Ethernet.

Check your ethernet cable, PC ethernet driver, and make sure your router ethernet port works.

Isaac96, 2017-10-23:
The resolution is set to 160x120 and the framerate is set at 2. The stream, however, is running at approximately 0.5 fps.

**The stream works,** it was a parental control setting that was blocking it. Thanks for the help!

closed 2017-10-23 by Isaac96