motioneye-project/motioneyeos #1198: Motion Detection Settings

HugoBerto, 2017-10-23:
I´m using motionEyeOS now for over a year and I´m still very happy with it, but I feel like I still haven´t understood the single variables which configurate the motion detection, since sometimes the camera won´t pick up obvious motion. I looked through some forums, but couldn´t find a clear answer.

These are my current motion detection settings: [http://fs5.directupload.net/images/171023/k72g72bd.png](url)

From my understanding the single variables do following:

- Frame Change Threshold: Set the minimum amount of frame changes in order for motion detection to occur
- Auto Noise Detection: If "On", don´t calculate noise into the frame change value
- Light Switch Detection: If "100%", don´t calculate fast changes of light into the frame change value
- Motion Gap: Gap in seconds between the current motion detection and the next motion detection
- Captured Before: Additional amount of frames that are saved before a motion detection
- Captured After: Additional amount of frames that are saved after a motion detection
- Minimium Motion Frames: Minimum amount of frames in which the configurated minimum amount of frame changes needs to occur.

I´d be very glad if someone could tell me if my understanding of the motion detection configuration is correct. If theres a useful topic with the same subject I´d be ok with the link too.

jasaw, 2017-11-14:
@HugoBerto Your understanding seems to be correct. This is the motion config file with some explanation: https://github.com/Motion-Project/motion/blob/master/motion-dist.conf.in.

closed 2017-11-14 by jasaw

drbeckma, 2017-12-27:
@HugoBerto , Could you reupload that picture of your current settings? I'm having a really rough go of tuning in my motion settings - it's either not detecting at all, or being far too sensitive. 

Thanks so much!

mhaddy, 2019-01-20:
@drbeckma You need to copy+paste the link text: http://fs5.directupload.net/images/171023/k72g72bd.png

zwep, 2019-10-17:
> @HugoBerto Your understanding seems to be correct. This is the motion config file with some explanation: https://github.com/Motion-Project/motion/blob/master/motion-dist.conf.in.

This link does not work anymore.. do you have an alternative? 

I see that you could also click a question mark in the UI.. but that doesnt work for me.. and I cant anywhere else any info on what the parameters mean (e.g. Despeckle filter)

jasaw, 2019-10-17:
Motion guide
https://motion-project.github.io/motion_guide.html

Detailed description of all the config items: 
https://motion-project.github.io/motion_config.html#Configuration_OptionsDetail

HJSK, 2020-04-30:
thanks a lot jasaw, that's exactly what i was looking for. Much more detailed then expected!