motioneye-project/motioneyeos #1199: Random Reboots

PotatoFi, 2017-10-24:
With the official Raspberry Pi camera connected to my Pi Zero W, the system randomly reboots. When connected to a monitor, I observed a kernel panic one time, twice I have observed a strange rainbow pattern on the bottom of the screen before it rebooted, but usually it just reboots.

If I disconnect the camera, the system is much more stable.

I have tried a different known-good power supply that happily powers my Pi2 and Pi3. 

Please let me know what log files are needed.

PotatoFi, 2017-10-24:
Switching the GPU memory to 96, the overclock setting to "PiZero", and disabling as many services as possible (such as Samba and FTP), and setting the video input to 320x200@15fps seems to have stabilized it.

jasaw, 2017-10-24:
Are you running out of memory and saw kernel OOM message?
Try setting "Captured Before" to 1 or zero because that takes up a lot of memory (all the frames are stored in memory).
What's your configuration?
Is "Movie" enabled? Which movie format did you choose?
What resolution and frame rate did you set to?

PotatoFi, 2017-10-24:
> Are you running out of memory and saw kernel OOM message?
I am not sure. Is there a way I can tell, perhaps a log file? Usually, there is no text output, just a hard reboot. There has been text output a couple of times, but it's gone too quickly for me to grab a photo.

> Try setting "Captured Before" to 1 or zero because that takes up a lot of memory (all the frames are stored in memory).
It was set to 1, so I changed it to 0.

> Is "Movie" enabled? Which movie format did you choose?
It is. I have H.264 selected. Is there a format that would be less RAM intensive?

> What resolution and frame rate did you set to?
320x200@15fps

PotatoFi, 2017-10-24:
I just checked uptime... 13 minutes. It's still randomly panicking or otherwise rebooting.

RPiSpy, 2017-10-24:
Another cause for a reboot could be it is losing WiFi connection. Try increasing the network timeout settings. My Pi Zero W system seems to reboot a lot and I'm sure it is due to it losing the WiFi signal.

jasaw, 2017-10-24:
There's no swap file or swap partition because that would wear the flash very quickly, so when the kernel is out of memory, it starts killing process and dump OOM message. In this case, motion process would be killed. You can see this OOM message if you SSH in and type `dmesg`. I'm not sure if it shows up in a log file. However, this shouldn't cause the device to reboot.

Which version of motionEyeOS are you running?

With your configuration, I don't think the device is running out of memory.

Try:
- using a different USB cable. Bad or low quality USB cable can cause device to reboot randomly too.
- turning off network timeout. This is to rule out reboots caused by wifi connectivity issue.
- reflashing the SD card again. Sometimes image is not flashed properly.

PotatoFi, 2017-10-24:
> Which version of motionEyeOS are you running?

20171003

> using a different USB cable. Bad or low quality USB cable can cause device to reboot randomly too.

I'm using a known good power adapter from Canakit. Tested it with another Pi, no problems. Also tested another known good power adapter with this Pi. I'm using a Raspberry Pi camera.

> turning off network timeout. This is to rule out reboots caused by wifi connectivity issue.

I have it set to 120 seconds. It should never take more than 10 seconds to disassociate and reassociate to my wireless network. I've run packet captures to confirm this.

> reflashing the SD card again. Sometimes image is not flashed properly.

I've done this as well, twice. I wonder if the SD card is bad? I would think that it wouldn't boot if that was the case.

Is anyone else using MotionEyeOS on a Raspberry Pi W with a Raspberry Pi camera?

jasaw, 2017-10-25:
I have multiple Pi Zero W with Pi Camera running motionEyeOS 20171008. Haven't got any problem.

> I wonder if the SD card is bad?

That's a possibility.
I've encountered a bad SD card once where certain files got corrupted. It booted fine, but crashed only when reading the corrupted files.

ccrisan, 2017-10-25:
The messages and dmesg log files should give us more details.

PotatoFi, 2017-10-25:
Upgraded to MotionEyeOS 20171008. Uptime this morning was 6 hours (should have been more like 10).

> The messages and dmesg log files should give us more details.

Sure thing, here they are:

[messages.log](https://github.com/ccrisan/motioneyeos/files/1414694/messages.log)

`Welcome to camera!
[root@camera ~]# uptime
 07:13:26 up  6:43,  load average: 0.37, 0.33, 0.35
[root@camera ~]# dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.9.41 (ccrisan@supermicro) (gcc version 4.9.4 (Buildroot 2017.02-gb87ca7a) ) #1 Mon Oct 9 00:19:38 EEST 2017
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt:Machine model: Raspberry Pi Zero W Rev 1.1
[    0.000000] cma: Reserved 8 MiB at 0x17400000
[    0.000000] Memory policy: Data cache writeback
[    0.000000] On node 0 totalpages: 98304
[    0.000000] free_area_init_node: node 0, pgdat c091abf0, node_mem_map d7c8be00
[    0.000000]   Normal zone: 864 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 98304 pages, LIFO batch:31
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 97440
[    0.000000] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:91:E1:D6 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.fiq_fix_enable=1 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait ro rootflags=noload panic=10 quiet loglevel=1 ipv6.disable=1
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 370952K/393216K available (5966K kernel code, 490K rwdata, 1956K rodata, 396K init, 733K bss, 14072K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xd8800000 - 0xff800000   ( 624 MB)
                   lowmem  : 0xc0000000 - 0xd8000000   ( 384 MB)
                   modules : 0xbf000000 - 0xc0000000   (  16 MB)
                     .text : 0xc0008000 - 0xc05dbdc8   (5968 kB)
                     .init : 0xc0847000 - 0xc08aa000   ( 396 kB)
                     .data : 0xc08aa000 - 0xc0924a98   ( 491 kB)
                      .bss : 0xc0924a98 - 0xc09dc288   ( 734 kB)\x008 - 0xc09dc288   ( 734 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000030] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000060] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000152] bcm2835: system timer (irq = 27)
[    0.000614] Console: colour dummy device 80x30
[    0.000639] console [tty1] enabled
[    0.000665] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
[    0.060301] pid_max: default: 32768 minimum: 301
[    0.060722] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.060738] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.061810] Disabling cpuset control group subsystem
[    0.062019] CPU: Testing write buffer coherency: ok
[    0.062079] ftrace: allocating 21793 entries in 64 pages
[    0.180778] Setting up static identity map for 0x8200 - 0x8238
[    0.182750] devtmpfs: initialized
[    0.191934] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[    0.192352] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.192378] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.193646] pinctrl core: initialized pinctrl subsystem
[    0.195069] NET: Registered protocol family 16
[    0.197680] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.207336] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
[    0.207352] hw-breakpoint: maximum watchpoint size is 4 bytes.
[    0.207456] Serial: AMBA PL011 UART driver
[    0.210224] bcm2835-mbox 2000b880.mailbox: mailbox enabled
[    0.211056] uart-pl011 20201000.serial: could not find pctldev for node /soc/gpio@7e200000/uart0_pins, deferring probe
[    0.259634] bcm2835-dma 20007000.dma: DMA legacy API manager at d880d000, dmachans=0x1
[... 119 lines cut ...]
[    1.190341] VFS: Mounted root (ext4 filesystem) readonly on device 179:2.
[    1.191398] devtmpfs: mounted
[    1.192639] Freeing unused kernel memory: 396K (c0847000 - c08aa000)
[    1.192646] This architecture does not have kernel memory protection.
[    1.200282] mmc1: new high speed SDIO card at address 0001
[    2.193707] EXT4-fs (mmcblk0p3): recovery complete
[    2.203246] EXT4-fs (mmcblk0p3): mounted filesystem with ordered data mode. Opts: (null)
[    2.478146] media: Linux media interface: v0.10
[    2.510511] Linux video capture interface: v2.00
[    2.570744] bcm2835-v4l2: scene mode selected 0, was 0
[    2.571394] bcm2835-v4l2: V4L2 device registered as video0 - stills mode > 1920x1200
[... cut ...]
[    4.214779] gpiomem-bcm2835 20200000.gpiomem: Initialised: Registers at 0x20200000
[    4.767429] usbcore: registered new interface driver brcmfmac
[    4.885773] brcmfmac: Firmware version = wl0: May 27 2016 00:13:38 version 7.45.41.26 (r640327) FWID 01-df77e4a7
[    6.137055] brcmfmac: power management disabled
[    6.405356] brcmfmac: power management disabled
[    8.083253] IPv6: Loaded, but administratively disabled, reboot required to enable
[    9.063400] system date is 2017-10-25 00:30:20
[  456.675170] random: crng init done
[root@camera ~]# 
`

jasaw, 2017-10-26:
This is the interesting bits I see in the messages.log.
> Oct 24 20:30:17 camera user.notice motioneye: not responding for 120 seconds, rebooting
Oct 24 20:30:24 camera user.notice motioneye: not responding for 120 seconds, rebooting
Oct 24 22:15:08 camera auth.info sshd[11038]: Disconnected from 192.168.0.115 port 62678
Dec 31 17:00:08 camera daemon.info dhclient: DHCPREQUEST on wlan0 to 255.255.255.255 port 67
Dec 31 17:00:08 camera daemon.info dhclient: DHCPACK from 192.168.0.1
Dec 31 17:00:08 camera daemon.info dhclient: bound to 192.168.0.135 -- renewal in 16640 seconds.
Oct 25 00:30:20 camera cron.info crond[793]: crond (busybox 1.26.2) started, log level 8
Oct 25 07:13:24 camera auth.info sshd[11507]: Accepted none for admin from 192.168.0.115 port 64183 ssh2

From this, I can deduce that:
At Oct 24 20:30:17, your Pi lost network connection, and rebooted. That's OK.
At Oct 24 22:15:08, you logged off from SSH.
At Oct 25 07:13:24, you logged in via SSH. It should have an uptime of 10 hours 45 minutes, but the uptime is only 6 hours 45 minutes.

The 6 hours uptime is because your Pi rebooted at Oct 25 00:30:20. From the logs, I can't see why it rebooted.

Another weird thing is every once in a while, you get DHCP request with the exact same timestamp of Dec 31 17:00:08.
> Dec 31 17:00:08 camera daemon.info dhclient: DHCPREQUEST on wlan0 to 255.255.255.255 port 67

ccrisan, 2017-10-26:
> Oct 24 20:30:17 camera user.notice motioneye: not responding for 120 seconds, rebooting

This indicates that motionEye does not respond for a long time. It means, in turn, that motion does not respond for a long time, thus, blocking the server. This is usually caused by a faulty camera and there's nothing motionEye can do about it.

closed 2017-10-26 by ccrisan

PotatoFi, 2017-10-26:
Thanks @ccrisan, I'll try a different cable, and then a different camera module. Thanks for your work, MotionEyeOS is very cool.

PotatoFi, 2017-11-02:
I have replaced the camera, and the exact same problem persists.

@ccrisan, could this be related to #1089 and #1072?