motioneye-project/motioneyeos #1202: Change Camera order

sorriso93, 2017-10-24:
Hello
I have 4 cameras connected to my raspberry and I would like to change camera order (camer id 1 to become cameraid 4)
Is that possible? From the GUI camera id is not editable
thanks

ccrisan, 2017-10-26:
Not yet possible, but planned.

Darthmonkey, 2018-05-06:
Any idea when this is planned?

IAmOrion, 2018-06-19:
+1 on this.  I have 4 RPi Zero's, and using a RPi 3 with 1TB HDD as a "hub" .... but as someone with OCD it really bugs me that I can't order the cameras.  I've named them incrementally and their IP's are also incremental.  Despite adding Camera 1 first, then Camera 2, Camera 2 still shows first.  I tried deleting them all, then added added them in reverse order... but Camera 2 is still showing as the 1st camera.

Chiny91, 2018-06-19:
Well... it depends how brave you are...   it looks as though you could simply edit the /data/etc/thread-n files, then reboot.   It could work, although if your R Pi 3 bursts into flames, it is all your fault 😄

IAmOrion, 2018-06-19:
@Chiny91 - Funny you should say that..... I literally just renamed them to the order I wanted.  It wasn't working at first, so went to get some food... just come back and saw your reply, then smacked my self in the head for not rebooting!! haha - just rebooted and all are in the order I wanted now :)

McMoobud, 2018-07-19:
@Chiny91 - Can you advise on which ID to change to do this ? I'm more than willing to give it a go, but when I switch the ID Tags around - I get duplicate cameras :\

Would appreciate advice as I'd love to switch mine around a bit !

McMoobud, 2018-07-19:
Thanks @IAmOrion ! 

I've done that, but it would appear it's not done anything which is odd. 

Might be how I've got mine configured or setup. I'll detail : 

Camera 1 - Garden View 
Raspi2 with Logitech Webcam 
Setup as the main MotionEye instance with the second camera attached as a MotionEye Remote Camera: 
Camera 2 - Front Door  
Raspi Zero W with 5mp Raspi Camera module 
Added remotely to the first camera. 

Maybe because I'm looking at Camera 1's interface directly, I could just switch and add Camera 1 to Camera 2's interface? 

I'll try that now :D 

Thanks again for your help, much appreciated!

CbsJVBJQ, 2018-09-30:
This worked for me on motionEyeOS 20180627

`ssh admin@x.x.x.x
cd /data/etc
ls -la thread-*
mv thread-4.conf thread-7.conf
`
reboot

Johnny-McCode, 2019-03-05:
> This worked for me on motionEyeOS 20180627. For example:
> 
> `ssh admin@x.x.x.x`
> `cd /data/etc`
> `ls -la thread-*`
> `mv thread-2.conf thread-10.conf`
> `mv thread-1.conf thread-2.conf`
> `mv thread-10.conf thread-1.conf`
> `reboot`

Thanks, worked for me perfectly, once rebooted I did have to turn "Video Device" On for all the cameras that I'd moved.

RichardRadford, 2019-09-22:
Would be nice to have as I'm always moving cameras around

ctrl50, 2020-03-10:
looking for a solution as well.. /data/etc doesnt exist on my setup.

starbasessd, 2020-03-10:
In newer versions the file names changed from 'thread-X.conf to camera-X.conf.
If you don't have /data/etc, try looking in /etc/motioneye (assuming you are only using motionEye, and not motionEyeOS)

PProvost, 2020-08-05:
Any updates on configuring this in the UI?

Barry-IA, 2020-10-02:
On a RPi4 I simply swapped the USB cables for my two camera around(!)  Both cameras into the USB 2.0 ports so just flip-flopped.  Yes, shocked me was that simple!  Hope this trick works for others.

Ruko2010, 2021-02-24:
Hi, I don't have MotioneyeOS installed, but motioneye as a package. Can you tell me where I can find the /data/etc directory to rename the cameras?

starbasessd, 2021-02-24:
The files are in /etc/motioneye on a motion/motionEye install on another OS.
The support you are looking for, @Ruko2010 is at
https://github.com/ccrisan/motioneye/issues

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

AlgeaPool, 2021-06-21:
I've been trying to get my trusty Camera1 to become Camera 11 so i can move it to the bottom of my camera list on the server. 

cd /etc/motioneye
sudo mv camera-1.conf camera-11.conf
sudo reboot 0

tried this on the server and it does not work

also tried it separately on the box the camera-1 runs on. neither way worked

any ideas, corrections, updates, anything will help. thanks

starbasessd, 2021-06-21:
@AlgeaPool are you using motionEye (on another OS, like RaspberryPiOS or Ubuntu) or motionEyeOS? 
It makes a difference as the files are located in different locations between the 2 of them.

starbasessd, 2021-06-21:
@AlgeaPool 
Also, within the camera-x.conf file there are a couple of settings to change:
# @id 1 
stream_port 8081
for your example, you'd want to change the first from id 1 to id 11 and the 2nd from 8081 to 8091 in case you decide to add/modify/delete other cameras to maintain consistency.

AlgeaPool, 2021-06-21:
raspbian with Motioneye.

AlgeaPool, 2021-06-21:
ok, i will move to Motioneye. but, while i still got you on this.....

you said above...._Also, within the camera-x.conf file there are a couple of settings to change:_

by '_with-in file_' you mean; open file and edit it?

But, sudo nano /camera-1.conf, produces new blank file (with no editable settings inside)

starbasessd, 2021-06-21:
if you do ls -la /etc/motioneye what is listed?
if you are looking to modify camera 1, you would use
sudo nano /etc/motioneye/camera-1.conf

AlgeaPool, 2021-06-21:
it works now. thanks.

closed 2021-06-21 by starbasessd